gyroflow/gyroflow #532: Can gyroflow realize real-time video stabilization？

GuoPingPan, 2022-12-10:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hello，i am looking forward real-time video stabilization algorithm.
Gyroflow has shown a great power in offline，can it be use in real-time？
If there is any document or advise for me to make secondary development？

Thank you very much!

AdrianEddy, 2022-12-10:
Not at this time, but it is on the roadmap at some point

AdrianEddy, 2022-12-10:
If you would like to help, feel free to message me on Discord so we can discuss it https://discord.gg/WfxZZXjpke

GuoPingPan, 2022-12-12:
I am very glad to receive your reply, and I sincerely hope to have the opportunity to assist you in your work!

closed 2022-12-12 by GuoPingPan