gyroflow/gyroflow #533: Gyroflow 1.3 and 1.2 not open on windows 10

longcaopr, 2022-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3

### What operating system are you using?

windows 10 buid 19044

### What GPU are you using?

RTX 2060

### What happened?

I try both version 1.2 and 1.3 but it won't open on my both computers. Please help!

### Relevant log output

```shell
19:27:08 [INFO] [MDK] MDK 0.18.0 (git 566d61a) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1934; MSVCRT14.34.31933.0/msvcp140.dll; 15:44:03 Dec 14 2022
19:27:08 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.1
19:27:08 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
19:27:08 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
19:27:08 [DEBUG] (1) mdk: [MDK] user module: C:\Users\ADMIN\Downloads\gyroflow\Gyroflow-windows64\Gyroflow.exe
19:27:08 [DEBUG] (1) mdk: [MDK] user module: C:\Users\ADMIN\Downloads\gyroflow\Gyroflow-windows64\Gyroflow.exe
19:27:08 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
19:27:08 [WARN] QQmlApplicationEngine failed to load component
19:27:08 [WARN] qrc:/src/ui/main_window.qml: No such file or directory
19:27:08 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 11.6.127") ["NVIDIA GeForce RTX 2060"]
19:27:08 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") Ok("OpenCL 3.0") ["Intel(R) UHD Graphics 770"]
19:27:08 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce RTX 2060
19:27:08 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
19:27:08 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
19:27:08 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
19:27:08 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 2060
```

AdrianEddy, 2022-12-14:
Workaround is here: https://github.com/gyroflow/gyroflow/issues/412#issuecomment-1201304649

longcaopr, 2022-12-15:
Could you a little more detail please. What does "download the code from the repository" and "green button code" mean? Where is the repository please.

AdrianEddy, 2022-12-15:
https://github.com/gyroflow/gyroflow

[image]

longcaopr, 2022-12-15:
I have copy UI folder to the same folder with gyroflow.exe. But it still not open. Here is the log:

04:27:20 [INFO] [MDK] MDK 0.17.0 (git 37eca09) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1933; MSVCRT14.33.31629.0/msvcp140.dll; 15:14:50 Oct 17 2022
04:27:20 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
04:27:20 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
04:27:20 [DEBUG] (1) mdk: [MDK] user module: C:\Users\ADMIN\Downloads\Gyroflow-windows64\Gyroflow.exe
04:27:20 [DEBUG] (1) mdk: [MDK] user module: C:\Users\ADMIN\Downloads\Gyroflow-windows64\Gyroflow.exe
04:27:20 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
04:27:20 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 11.6.127") ["NVIDIA GeForce RTX 2060"]
04:27:20 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") Ok("OpenCL 3.0") ["Intel(R) UHD Graphics 770"]
04:27:20 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce RTX 2060
04:27:21 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
04:27:21 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
04:27:21 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 2060
04:27:21 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
04:27:21 [DEBUG] (1) default: QT_QUICK_CONTROLS_TEXT_SELECTION_BEHAVIOR ""
04:27:21 [ERROR] thread 'main' panicked at 'index out of bounds: the len is 9 but the index is 9': src\ui\components\TimelineGyroChart.rs:76
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: QMetaObject::metacall
  12: QQmlObjectCreator::setupFunctions
  13: QQmlPropertyValueInterceptor::operator=
  14: QQmlPropertyValueInterceptor::operator=
  15: QV4::QObjectMethod::callInternal
  16: QV4::FunctionObject::call
  17: QV4::Value::toUInt16
  18: QV4::Value::toUInt16
  19: QV4::Function::call
  20: QQmlJavaScriptExpression::evaluate
  21: QQmlBinding::evaluate
  22: QQmlBinding::dependencies
  23: QQmlBinding::update
  24: QQmlBinding::setEnabled
  25: QQmlObjectCreator::finalize
  26: QQmlIncubatorPrivate::incubate
  27: QQmlIncubationController::incubateFor
  28: QQuickWindow::hideEvent
  29: QObject::event
  30: QCoreApplication::notify
  31: QCoreApplication::notifyInternal2
  32: QEventDispatcherWin32Private::sendTimerEvent
  33: QEventDispatcherWin32::event
  34: QCoreApplication::notify
  35: QCoreApplication::notifyInternal2
  36: QCoreApplicationPrivate::sendPostedEvents
  37: QWindowsGuiEventDispatcher::sendPostedEvents
  38: QEventDispatcherWin32::processEvents
  39: QWindowsGuiEventDispatcher::processEvents
  40: QEventLoop::exec
  41: QCoreApplication::exec
  42: <unknown>
  43: <unknown>
  44: <unknown>
  45: <unknown>
  46: BaseThreadInitThunk
  47: RtlUserThreadStart

04:27:21 [WARN] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
04:27:21 [WARN] "There are still \"2\" items in the process of being created at engine destruction."

AdrianEddy, 2022-12-15:
Use latest dev build from here https://gyroflow.xyz/devbuild/?autodownload and copy that UI folder there

longcaopr, 2022-12-15:
Yes that works. Thank you very much. Have a nice day Adrian ;)

AdrianEddy, 2022-12-22:
@longcaopr do you use any additional antivirus software?

AdrianEddy, 2022-12-22:
Could you please try the latest dev build and let me know if it shows the same error? https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-12-22:
@longcaopr @gurfing @underlines  Could any of you please try downloading latest dev build, then adding this line to `Gyroflow_with_console.bat` as a second line:
`set QT_LOGGING_RULES=*.debug=true`
then running this `Gyroflow_with_console.bat` file.

Then close Gyroflow (kill in task manager) and upload gyroflow.log here

tootk90, 2023-01-12:
> 

I downloaded the last devbuild and gyroflow is not starting. 
The strange thing is that the first time I downloaded the stable version, it started and worked with no issue. After I restarted the computer the UI is not starting. 

I downloaded the zip file with the source code, and copied the ui folder in the same folder as Gyroflow.exe. Still not working

here is the gyroflow.log from the last dev version

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10406579/gyroflow.log)

AdrianEddy, 2023-01-12:
@tootk90 do you see the application window, or no window at all? Your issue looks like GPU driver issue.

Please install these GPU drivers for NVIDIA: https://www.nvidia.com/Download/driverResults.aspx/198401/en-us/
and for Intel: https://downloadmirror.intel.com/764512/gfx_win_101.2115.exe

tootk90, 2023-01-12:
> @tootk90 do you see the application window, or no window at all? Your issue looks like GPU driver issue.
> 
> Please install these GPU drivers for NVIDIA: https://www.nvidia.com/Download/driverResults.aspx/198401/en-us/ and for Intel: https://downloadmirror.intel.com/764512/gfx_win_101.2115.exe

updated the gpu drivers. Didn't work.
Connected an external monitor (this morning i was connected to the external monitor when i  tried it for the first time), and the program started. 

Disconnected the monitor, the program still works

AdrianEddy, 2023-01-12:
ah so it probably remembered the program placement on the other monitor, so it was not visible on that one monitor.

In this case you can Shift+Right click on an icon in taskbar and click Maximize. 
Or remove settings in regedit from `Computer\HKEY_CURRENT_USER\Software\Gyroflow\Gyroflow`

AdrianEddy, 2023-01-19:
@longcaopr @gurfing @underlines does any of you still have that problem in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2023-01-31:
Fixed in 89a394486884edde2f337bff0114dab924d1d4a3

closed 2023-01-31 by AdrianEddy