gyroflow/gyroflow #534: v1.3 can't open certain file on Intel Mac Pro

AfflatusX, 2022-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.3 (dev and prod)

### What operating system are you using?

macOS Monterey

### What GPU are you using?

AMD Radeon Pro 580X

### What happened?

When trying to load video files with YUV422 10bit HEVC format, Gyroflow just shows the spinner forever without loading the video.

I noticed this is only happening on Intel Mac Pro (Xeon W + 580X Pro), but not happening on my M2 Macbook Air. Unfortunately Air's GPU doesn't seem to support the export encoding.

Is it an known issue that this format is not supported?

AfflatusX, 2022-12-18:
Btw I think it might have something todo with gamma / color profile, the footage was shot using Sony a7s iii, Gamma S-Log3, Color Mode: S-Gamut3.Cine.

I tried to use a different file without the color profile and it seems to work on both intel and M1 mac.

AfflatusX, 2022-12-18:
You can use the sample video I uploaded to Dropbox in the issue description to test it out.

AdrianEddy, 2022-12-18:
I can't reproduce it on Windows or on M2 Air, and I don't have Intel Mac to test. 
On your Intel mac, after the video fails to load, close gyroflow, go to "Show package contents -> Contents -> MacOS" and upload `gyroflow.log` file from there.

Also, what exactly do you mean by `Unfortunately Air's GPU doesn't seem to support the export encoding.`?
M2 has a dedicated hardware media engine and supports way more accelerated exporting formats than Intel+AMD.

AfflatusX, 2022-12-18:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10254183/gyroflow.log)

log attached.

So the footage was shot with S-log3 - so in order to preserve the light/color information, I either do a color grading before export, stablize it, then export to other format (that's what I do with Sony Catalyst Browse), or, run gyroflow and export in the same format so that I can post process it in Premiere.

With Gyroflow v1.3, when I click export, it tells me exporting as original format isn't supported (see screenshot below). I tried the other format it suggested but they mostly lose all the pixel information (because S-LOG3 is over exposed without LUT applied).

That's the reason I need to use the mac pro to do the export - since it has 32 cores on CPU.

[image]

wang-bin, 2022-12-20:
> @wang-bin looks like the pixel format is not supported, can it be added or can we do a fallback in this case?

yes. the format is almost the same as m1/2, except color range is different. i will fix it today.

wang-bin, 2022-12-21:
@AdrianEddy try the last build

AdrianEddy, 2022-12-22:
@AfflatusX  please try latest dev build https://gyroflow.xyz/devbuild/?autodownload and post gyroflow.log if it's still not working

AfflatusX, 2022-12-22:
yup can confirm it's working now! Thanks for the fix!

AfflatusX, 2022-12-22:
actually before I close - @AdrianEddy / @wang-bin seems like neither 580X, Apple M2, or 3080Ti supports YUV422P10LE format for encoding. 

I'm wondering is that a HW issue or a SW issue?

AdrianEddy, 2022-12-22:
YUV formats are not really supported by apple, they are coming from software decoders. Make sure to check `Use GPU decoding` under Advanced (or uncheck and re-check it), then it should use P210LE, which is equivalent to YUV422P10LE 
P010LE is equivalent to YUV420P10LE

AfflatusX, 2022-12-22:
Cool thanks for the answer!

closed 2022-12-22 by AfflatusX

AfflatusX, 2022-12-22:
@AdrianEddy  It doesn't look like P210LE was given as an alternative px format - see log:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10282968/gyroflow.log)

I noticed these two lines: 

01:19:06 [DEBUG] (1) gyroflow::rendering: Possible encoders with Amd: [("hevc_videotoolbox", true), ("libx265", false)]

and 

01:19:06 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [VIDEOTOOLBOX, NV12, YUV420P, BGRA, P010LE]

AdrianEddy, 2022-12-22:
Did you try on M2? Intel devices had much more limited media processing engine in the hardware

AfflatusX, 2022-12-22:
Yeah same results on M2. However, when choosing ProRes HQ the option showed up. I guess HEVC + P210LE isn't a valid combo?

wang-bin, 2022-12-22:
My m1 air supports main42210, maybe p210 is a supported format

```
fcc: 'hvc1' -  Apple HEVC (HW)  hw=1 {
    CodecName = HEVC;
    CodecType = 1752589105;
    DisplayName = "Apple HEVC (HW)";
    EncoderID = "com.apple.videotoolbox.videoencoder.ave.hevc";
    EncoderName = "Apple HEVC (HW)";
    IsHardwareAccelerated = 1;
    PerformanceRating = 400;
    QualityRating = 90;
    SupportedSelectionProperties =     {
        ProfileLevel =         (
            "HEVC_Main_AutoLevel",
            "HEVC_Main10_AutoLevel",
            "HEVC_Main42210_AutoLevel"
        );
    };
}
```

AdrianEddy, 2022-12-22:
Hmm, doesn't look like ffmpeg supports that profile?
https://github.com/FFmpeg/FFmpeg/blob/eef763c7057a7f5f4b7dae7855d07b2a6da8b537/libavcodec/videotoolboxenc.c#L2775-L2785

wang-bin, 2022-12-22:
ffmpeg needs a patch to support p210. I've confirmed on my m1

ntthaibk, 2024-07-11:
sorry to bring this up, I'm using v1.5.4 on macbook pro Apple M1, my video format is YUV422P10LE from Sony a6700

 but when checking for p210, there's none under h265 hevc, other pixel format supported is either 420 10 bit or 420 8 bit.
[image]

tried with h264 it's work but much slower rendering. Prores also worked but file size become a whooping 1.36Gb compare to 200 mb original file

AdrianEddy, 2024-07-11:
ffmpeg doesn't support 4:2:2 10-bit rendering, even though the chip does. We're waiting for ffmpeg to integrate this. In the meantime your only option to render 4:2:2 10-bit is to render on CPU, or record 4:2:0 10-bit in camera, or use the OpenFX plugin in Resolve

ntthaibk, 2024-07-11:
> ffmpeg doesn't support 4:2:2 10-bit rendering, even though the chip does. We're waiting for ffmpeg to integrate this. In the meantime your only option to render 4:2:2 10-bit is to render on CPU, or record 4:2:0 10-bit in camera, or use the OpenFX plugin in Resolve

thank you @AdrianEddy , just wonder out of the supported pixel format listed in hevc, what should I pick from, checked in Catalyst browse after export, I noticed only P10LE is 4:2:0 10 bit, but will it lose a lot of information compare to directly capture the footage 4:2:0 10 bit out of the camera?

AdrianEddy, 2024-07-11:
P010LE is 4:2:0 10-bit, and you can use the same in gyroflow. There's not much difference between 4:2:0 and 4:2:2 if you're not doing any greenscreen work. It's better to capture 4:2:0 10-bit in camera, but converting 4:2:2 to 4:2:0 is fine too

ntthaibk, 2024-07-11:
thank you a lot! @AdrianEddy , but eventually I will give prores proxy a try since it's not so much heavier :D :D :D

AdrianEddy, 2024-07-11:
@ntthaibk Try 4:2:2 in latest dev build, make sure you have both GPU decoding and encoding checked

ntthaibk, 2024-07-27:
hi @AdrianEddy , sorry for the late reply, I've tried 4:2:2 with latest dev build but still got this message.
I'm on Mac M1

[image]

[image]

this is the file specs

[image]