gyroflow/gyroflow #535: Do at least one pre-release

lukegluke, 2022-12-19:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Dear @AdrianEddy,

Thanks for you great job in this project! But could you please in future make at least one prelease before new release? For example I'm not able to contribute much into code, so I don't follow project carefully, but I'm glad to contribute to translation. I'm subscribed to new releases and it will be nice to see RC to be able to translate new strings before main release.

AdrianEddy, 2022-12-19:
There will be next minor release soon, this one was put out there to be available once new DJI action 2 firmware with Gyroflow support releases tomorrow from DJI.

You can follow discord, i often mention the release beforehand there in general channel.

closed 2022-12-19 by AdrianEddy

AdrianEddy, 2022-12-22:
I'm planning to release v1.4.1 tomorrow so if you want to update translations feel free to do it by then