gyroflow/gyroflow #541: Add Type Identifiers for the .gyroflow file format

latenitefilms, 2022-12-21:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Please considering adding a Type Identifier for the `.gyroflow` file format on macOS. Once this is done, it would be great if you could drag and drop `.gyroflow` files onto the Gyroflow Dock Icon. It would also be great if you could "Open With..." the Gyroflow app.

latenitefilms, 2023-01-18:
I've started this in #571.

Note to self - @AdrianEddy says:

> https://doc.qt.io/qt-6/qfileopenevent.html
> this will get tricky, as the QApplication is wrapped in qmetaobject-rs crate, and not on my side
> but, installEventFilter can make it easy https://doc.qt.io/qt-6/qobject.html#installEventFilter
> this will all need to be in C++

AdrianEddy, 2023-09-21:
This is now implemented on all platforms except Linux, where we don't have proper desktop integration anyway, and the user base and interest is too small to be worth the effort. 
On Windows, this is implemented in .appx (app installed from windows store) which is not published yet, but will be in near future

closed 2023-09-21 by AdrianEddy