gyroflow/gyroflow #544: Unable to export video on a MAC after updating to 1.4

Phiqtional, 2022-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4

### What operating system are you using?

MacOS Monterey 12.6

### What happened?

Tried to export video but an error popped up. 

Never had any issues on 1.3 only started experiencing this right after updating to 1.4

This is the error that popped up. 

[image]

Any help would be greatly appreciated. Thank you.

AdrianEddy, 2022-12-22:
It's already fixed https://gyroflow.xyz/devbuild/?autodownload

closed 2022-12-22 by AdrianEddy

Phiqtional, 2022-12-22:
Thank you so much AdrianEddy!