gyroflow/gyroflow #545: Export not working: ffmpeg / Nvenc version error

PaWi20, 2022-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4

### What operating system are you using?

Windows 10

### What GPU are you using?

NVIDIA GeForce GTX 1650 Ti

### What happened?

I updated to version 1.4. Now, it's not possible to stabilize the footage anymore in H.265 with GPU encoding. It works without GPU encoding (H.265) and H.264 (with GPU encoding). 

I'm not sure if it worked in H.265 with GPU encoding in a previous version. But I assume this, as I didn't change the settings after the upgrade.

You can see the errors in the attached screenshot:
[image]

PaWi20, 2022-12-22:
I'm not sure how to provide a log file.

mdj36, 2022-12-22:
the same error nvidia geforce gtx 1060 the same error please fix earlier was pretty well working.

AdrianEddy, 2022-12-22:
Did you try different `Device for rendering`? It's in Export settings -> Advanced. Also please provide logs after rendering from both 1.3 and 1.4 (gyroflow.log file is created next to Gyroflow.exe after you close Gyroflow) if it doesn't, run Gyroflow as administrator

mdj36, 2022-12-22:
I used 1.3rc it was working fine.:
[image]
[image]

On 1.4 with same settings:
[image]
[image]

AdrianEddy, 2022-12-22:
Please upload logs

mdj36, 2022-12-22:
1.4 not working:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10286858/gyroflow.log)

mdj36, 2022-12-22:
1.3 working:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10286958/gyroflow.log)

AdrianEddy, 2022-12-22:
Should be fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-12-22 by AdrianEddy

PaWi20, 2022-12-23:
Thanks a lot, also to @mdj36 for providing the log files. I was really busy yesterday and couldn't reply.