gyroflow/gyroflow #547: Gyroflow crashed on file export

dezcom, 2022-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

2022-12-23 night build

### What operating system are you using?

OS Version:            Mac OS X 10.15.7 (19H2026)

### What GPU are you using?

intel 4000

### What happened?

Applica tion crashed on exporting of the dji avata file

### Relevant log output

```shell
Log file
https://disk.yandex.ru/d/AVZdEJJ_4S6G0g
```

AdrianEddy, 2022-12-23:
Please change `Device for video processing` under Advanced to one starting with `[wgpu]` and let me know if that worked

dezcom, 2022-12-23:
that works, thank you

closed 2022-12-23 by AdrianEddy