gyroflow/gyroflow #548: Error when exporting GoPro Hero 11 footage

mikepostons, 2022-12-23:
Getting this error when exporting GoPro Hero 11 footage on a M1 Macbook.  Footage was shot in 4:3 at 4K 25fps.  GyroFlow settings below:

[image]

[image]

AdrianEddy, 2022-12-23:
It's already fixed https://gyroflow.xyz/devbuild/?autodownload

closed 2022-12-23 by AdrianEddy

mikepostons, 2022-12-23:
Ah amazing.  Downloaded the latest version just before and didn't see the issue/resolution on GitHub.  Thanks