gyroflow/gyroflow #549: Rendering time significantly longer on macOs Ventura

butchmarzee, 2022-12-23:
Same file, same settings, same gyroflow 1.4.2 release, with macOs Monterey 40 frame/sec, with Ventura 30 frame/sec.
Apple silicon m1 max

AdrianEddy, 2022-12-23:
Please try different "Device for video processing" under Advanced and let me know the results

AdrianEddy, 2022-12-23:
Also let me know if you have "Use GPU decoding" (also under Advanced) enabled

butchmarzee, 2022-12-23:
> Please try different "Device for video processing" under Advanced and let me know the results

Tried wgpu , result is pretty much the same, few fps lower

butchmarzee, 2022-12-23:
> Also let me know if you have "Use GPU decoding" (also under Advanced) enabled

Just tried to turn it off, and fps went up to 41-42! Shouldn’t work the other way?

AdrianEddy, 2022-12-23:
it depends, rendering is complicated. Can you get me gyroflow.log from:
1. Ventura GPU decoding enabled
2. Ventura GPU decoding disabled
3. Monterey GPU decoding enabled
4. Monterey GPU decoding disabled

Show package contents -> Contents -> MacOS -> gyroflow.log

Start rendering, wait a few seconds, cancel it and close gyroflow, then upload gyroflow.log

butchmarzee, 2022-12-23:
> it depends, rendering is complicated. Can you get me gyroflow.log from:
>
> 1. Ventura GPU decoding enabled
> 2. Ventura GPU decoding disabled
> 3. Monterey GPU decoding enabled
> 4. Monterey GPU decoding [...]

I can do with ventura, unfortunately just updated from monterey so that option is gone :(

butchmarzee, 2022-12-23:
> it depends, rendering is complicated. Can you get me gyroflow.log from:
>
> 1. Ventura GPU decoding enabled
> 2. Ventura GPU decoding disabled
> 3. Monterey GPU decoding enabled
> 4. Monterey GPU decoding [...]
[gyroflow_gpu_decoding_on.log](https://github.com/gyroflow/gyroflow/files/10297635/gyroflow_gpu_decoding_on.log)
[gyroflow_gpu_decoding_off.log](https://github.com/gyroflow/gyroflow/files/10297636/gyroflow_gpu_decoding_off.log)

AdrianEddy, 2022-12-23:
That's really weird, the `decoding_off` indicates there's extra pixel format conversion step (expected), which is not cheap so it should be much slower, and `decoding_on` has everything perfectly without extra steps according to the log

butchmarzee, 2022-12-23:
> That's really weird, the `decoding_off` indicates there's extra pixel format conversion step (expected), which is not cheap so it should be much slower, and `decoding_on` has everything perfectly without extra steps according to the log

That’s exactly what I thought, something is wrong here

butchmarzee, 2022-12-23:
> That's really weird, the `decoding_off` indicates there's extra pixel format conversion step (expected), which is not cheap so it should be much slower, and `decoding_on` has everything perfectly without extra steps according to the log

I have just tried with a new m2 macbook pro, similar results, gpu decoding off makes it much faster

butchmarzee, 2022-12-26:
> > That's really weird, the `decoding_off` indicates there's extra pixel format conversion step (expected), which is not cheap so it should be much slower, and `decoding_on` has everything perfectly without extra steps according to the log
> 
> That’s exactly what I thought, something is wrong here

I have done some more tests:
The earlier results were recorded with dji avata, today i tried with gopro hero 10, and with the gorpo it works as i reckon should. With gpu decoding rendering time is much shorter, so i believe there is a bug with the avata’s settings

AdrianEddy, 2022-12-26:
avata 8-bit or 10-bit?
gopro 8-bit or 10-bit?

butchmarzee, 2022-12-26:
> avata 8-bit or 10-bit?
> gopro 8-bit or 10-bit?

Both 8 bit

butchmarzee, 2022-12-26:
> > avata 8-bit or 10-bit?
> > gopro 8-bit or 10-bit?
> 
> Both 8 bit

Avata h264 recorded, gopro hevc recorded

closed 2025-09-04 by AdrianEddy