gyroflow/gyroflow #550: View R3D Footage

jim89457, 2022-12-26:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Would it be able to view the R3D clips directly from RED cameras instead of having to go through the extra step of converting them to a different format prior?

AdrianEddy, 2022-12-26:
Probably yes at some point

jim89457, 2022-12-26:
How can we get it sooner / accelerate it?

AdrianEddy, 2022-12-26:
See this answer for a bit more info how this works: https://github.com/gyroflow/gyroflow/issues/443#issuecomment-1229459504
Basically @wang-bin needs to implement that in mdk and he needs to get paid for it, so to accelerate it community needs to raise money to pay him for his work

jim89457, 2022-12-26:
How much money do you think for R3D?

danielhager, 2023-01-03:
Yes, please.
And please donate to them.

AdrianEddy, 2023-01-09:
R3D playback is coming soon, you can donate here:
https://donate.stripe.com/5kAaGv6ky5mU1s44gg
https://buymeacoffee.com/adrianeddy
https://paypal.me/AdrianEddy

AdrianEddy, 2023-01-18:
R3D playback is available in the latest dev build now https://gyroflow.xyz/devbuild/?autodownload

closed 2023-01-18 by AdrianEddy