gyroflow/gyroflow #552: DJI Action2 footage have some bad not smooth moves

AlexeyZatsepin, 2022-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4.2

### What operating system are you using?

macOS Ventura 13.0.1

### What GPU are you using?

AMD Radeon Pro 5300M 4 GB

### What happened?

https://user-images.githubusercontent.com/10632284/209819124-8abda077-acfc-41c5-abea-c8b5acc31e52.MOV

https://user-images.githubusercontent.com/10632284/209819145-3673fbec-b394-4fe9-87b6-9f6b230d95ea.MOV

AlexeyZatsepin, 2022-12-28:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10314421/gyroflow.log)

AdrianEddy, 2022-12-28:
was it 4k 50fps mode by any chance? can you share the original video file?

AlexeyZatsepin, 2022-12-28:
it's too big for github, how can I share it to you?

AdrianEddy, 2022-12-28:
wetransfer, dropbox, google drive, or mega.co.nz

AlexeyZatsepin, 2022-12-29:
internet is so bad, trying to upload

AlexeyZatsepin, 2022-12-31:
@AdrianEddy finally manage to upload here: https://fex.net/ru/s/ynn8vea

hzjianglf, 2023-01-01:
I find this bug，too，pls  see video： https://pan.baidu.com/s/1wKRk9OK5q8cQX2ecFncAkQ?pwd=4444    pwd: 4444

hzjianglf, 2023-01-01:
> [AlexeyZatsepin](/AlexeyZatsepin)

@AlexeyZatsepin

AdrianEddy, 2023-01-01:
it's a bug in DJI Action 2 firmware, I reported it to DJI.

There's nothing you can do about this file unfortunately, it will likely be fixed in the next firmware version, but this video will remain like that. You can add some small amount of Warp Stabilizer in Premiere or Resolve

AlexeyZatsepin, 2023-01-02:
Ah, unfortunately, thanks for figuring this out!

AdrianEddy, 2023-01-18:
This is fixed in latest Action 2 firmware - v01.04.0420

closed 2023-01-18 by AdrianEddy