gyroflow/gyroflow #558: Investigate removing GPL dependancies from Gyroflow Core

latenitefilms, 2023-01-01:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Currently `gyroflow_core` has a single GPL dependancy:

- [ahrs-rs](https://github.com/jmagnuson/ahrs-rs)

There is already an [issue](https://github.com/jmagnuson/ahrs-rs/issues/21) on [ahrs-rs](https://github.com/jmagnuson/ahrs-rs) discussing making this repo MIT licensed, but it hasn't been updated since April 2021.

@jmagnuson writes:

> Thanks for the interest, licensing was definitely something I grappled with when first writing. Ultimately, it comes down to the original source implementation (GPL) which may or may not have been referenced while porting. IANAL so I simply chose to stay on the safe side, though I have thought about reaching out to Dr. Madgwick for any sort of blessing that may be offered. FWIW this is the only crate I maintain which isn't MIT/Apache-2.0, so I totally agree with your point of view.

I've posted [this reply](https://github.com/jmagnuson/ahrs-rs/issues/21#issuecomment-1368543255) - as it looks like the IMU and AHRS algorithms might actually be MIT licensed now after all?

As discussed elsewhere (including #34), so that iOS, iPadOS and macOS App Store Apps can be created for Gyroflow, it would be great if `gyroflow_core` could be licensed as MIT, so that it can be used in App Store apps. The main Gyroflow app could remain GPL3.0 - it's just the "core" that's required.

AdrianEddy, 2023-01-01:
FYI, for iOS/iPad apps, the entire code will be used, not only gyroflow_core

AdrianEddy, 2023-01-01:
GPL dependencies can be removed, it's not a big problem.

The main issue for me is that I believe GPL prevents anyone to monetize this code and create closed-sourced derivatives.
I'm putting insane amounts of work, life and money (hundreds thousands of dollars of my personal money goes into this), and I really hate the idea that someone can take my work and create something non-open. What are other licenses that would prevent that but not be the GPL?

latenitefilms, 2023-01-01:
Ah, right - I just assumed for an iOS app you'd basically just build the UI again from scratch in Swift/Obj-C, and use AV Foundations for playback. Have you already started an iOS app repo?

AdrianEddy, 2023-01-01:
The main repo is for all platforms - Windows/Linux/MacOS/Android/iOS
iOS building steps are listed in the README, the code already builds fine, but I didn't have time to finish the last step which is packaging the compiled code (.a) to the app bundle (.ipa). I have some resources for that if you want to dig into this. Basically now we need to create an XCode project and call `main` + add all stuff like Info.plist, icons, permissions etc

latenitefilms, 2023-01-01:
> GPL dependencies can be removed, it's not a big problem.
>
> The main issue for me is that I believe GPL prevents anyone to monetize this code and create closed-sourced derivatives.
> I'm putting insane [...]

Totally understand - and I hugely appreciate all the insane amount of work you put into this awesome project. My main coding project, [CommandPost](https://commandpost.io) - is also free and open source, so I know first hand how much time goes into both development and support.

I guess one way to do it is keep everything GPL, and do something like this:

https://opensource.stackexchange.com/a/8682

Essentially, you keep your main repo GPL3.0, but you could manually grant yourself other 3rd parties another license for App Store distribution, under the proviso that that code is also open source on GitHub.

So, in the case of Gyroflow Toolbox, you could grant us a license to `gyroflow_core` under the proviso that all the code is open source on GitHub, so that I can attempt to post Gyroflow Toolbox on the Mac App Store - either for free or paid, depending on your preference. This solves the licensing issue - however, there's still technical issues to solve, such as  #557.

> The main repo is for all platforms - Windows/Linux/MacOS/Android/iOS
> iOS building steps are listed in the README, the code already builds fine, but I didn't have time to finish the last step which [...]

I've only ever worked on macOS apps thus far - but I'm definitely interested in having a look at this at some point. I feel like I now know enough about Xcode and Objective-C to be very dangerous, haha.

AdrianEddy, 2023-01-01:
> I guess one way to do it is keep everything GPL, and do something like this:
> https://opensource.stackexchange.com/a/8682

That's a nice option, we can do that.

mdk-sdk still remains a problem (it's closed source, and GPL requires all dependencies to be open-source), but it's already tracked in #381 and I have some ideas for that.

latenitefilms, 2023-01-01:
Sounds like a plan. If you're up for it then, lets try solve the sandbox and licensing issues, then we can try get Gyroflow Toolbox up on the Mac App Store as a proof of concept. I'm sure App Review will bring up unexpected issues, that we'll need to solve for iOS/iPadOS anyway.

I'm happy to keep Gyroflow Toolbox free on the Mac App Store, or we could charge a token amount (USD$5?) and split any profits however you want to help keep funding development and updates?

AdrianEddy, 2023-01-03:
Let's keep the license discussion in #381, no reason to have two threads open

closed 2023-01-03 by AdrianEddy

latenitefilms, 2023-01-03:
It looks like `ahrs-rs` will soon be MIT licensed anyway, so that's a win! That should mean all the code in `gyroflow_core` will be MIT/Apache or your own code, so you can add some license tweaks to Gyroflow to allow App Store usage.