gyroflow/gyroflow #559: Video from DJI Action2 exported as just black screen

AlexeyZatsepin, 2023-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4.2

### What operating system are you using?

macOS Ventura 13.0.1

### What GPU are you using?

AMD Radeon Pro 5300M 4 GB

### What happened?

Video from DJI Action2 exported as just black screen, by link you can see before and after gyroflow: 

https://fex.net/s/kndcnmv

AdrianEddy, 2023-01-03:
do you see the preview in Gyroflow or is it also black? Try resetting Gyroflow to defaults (Advanced -> Reset all settings to defaults, on bottom right)

AlexeyZatsepin, 2023-01-03:
also black, already tried to reset to defaults - not worked

AdrianEddy, 2023-01-03:
Please show me a screenshot of Gyroflow with your settings

AlexeyZatsepin, 2023-01-03:
[image]

closed 2023-01-03 by AdrianEddy