gyroflow/gyroflow #560: not able to creat a new lens correct file

ericdesigners, 2023-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.4.2

### What operating system are you using?

win10

### What GPU are you using?

1080ti

### What happened?

camera: sony fx3 3840*2160 @ 60fps stab off,
Lens: sony 14mm f 1.8 gm
error number is always 999.0

attach video files to download from baidu netdisk

LINK：https://pan.baidu.com/s/1XZQjF_D29LGePxLU8Nkp2w?pwd=qml1 
password：qml1

AdrianEddy, 2023-01-03:
How can I download from baidu without account?

Also did you try to increase Sharpness limit in Advanced?

AdrianEddy, 2023-01-09:
Do you still have the issue?

closed 2023-01-23 by AdrianEddy