gyroflow/gyroflow #562: doesn`t support pixel format(yuv420p10le)

Czukuru, 2023-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4.2

### What operating system are you using?

windows 11

### What GPU are you using?

GTX 760

### What happened?

I use my gopro 11 with 5.3k 8:7 10bit color and have this: "gpu accleated encoder does't support this pixel format(yuv420p10le)"
[image]
i chose all formats, but useless
[image]
[image]
with standard colors it`s work grate

AdrianEddy, 2023-01-07:
Also install latest drivers for your GPU

closed 2023-01-09 by AdrianEddy

Czukuru, 2023-01-13:
> Я ничего не могу с этим поделать в Gyroflow — ваш графический процессор не поддерживает 10-битное ускоренное кодирование HEVC. Вы должны рендерить без GPU или менять видеокарту на более новую. Кодирование ЦП медленное, но вы можете добавить несколько клипов в очередь рендеринга и оставить его на ночь.

I'm sorry, but i bought new rtx 2060super 8gb and it's still happening

AdrianEddy, 2023-01-13:
Did you reset Gyroflow settings to default? "Reset all settings to defaults" button in the Advanced panel on bottom right.
If it's still happening on RTX2060 after resetting gyroflow, please send me gyroflow.log file