gyroflow/gyroflow #563: DJI O3 video downsize a lot and Y rotate 180 degree.

Spielben, 2023-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4.2

### What operating system are you using?

Mac OS Monterrey 12.16

### What GPU are you using?

Intel UHD Graphics 630 1536 MB

### What happened?

When I drag or upload a video from DJI O3 Air Unit, at first all seems ok.
Gyro datas are found and all file info are correct.
Then when I toggle stabilisation the picture reduce more than X10 and rotate 180.
I tried 2 different video footage with the same issue. and when export the problem is the same.
I have shot without IES and in wide field of view.
[image]
[image]

thank you for your help

### Relevant log output

```shell
https://drive.google.com/file/d/1vtOYLnSQpf8iUr-8TG0EnxxJCAjulZDz/view?usp=sharing
```

AdrianEddy, 2023-01-06:
Did you try to reset all settings to defaults? (bottom right in Advanced panel)

B-nutze-RR, 2023-01-06:
The O3 clip shows,  a lot of vibrations make it into the camera ( and don't show up in video). You need to check if your RPM filters are configured correctly - using high PWM frequency for your ESCs is recommended.  You may also need a TPU mount for the camera.

Spielben, 2023-01-07:
That was the missing step.
The “Damned” reset option.
Thanks a lot,

Cheers,

Ben SPIELBEN

Spielben & Co.

✆ +66 91790 44 41
***@***.***
w: https://spielben.com

closed 2023-01-07 by AdrianEddy

maxartgym, 2023-02-06:
Please I have the same problem with my djo ao3, did you resolve this issue, explain me please what did you reset to the Solution.... Thanks alot

AdrianEddy, 2023-02-07:
> Please I have the same problem with my dji ao3, did you resolve this issue... , explain me pleaseee what did you exactly do for Solution...Thanks alot...

Open Advanced panel on bottom right and click "Reset all settings to defaults"

maxartgym, 2023-02-07:
> > Please I have the same problem with my dji ao3, did you resolve this issue... , explain me pleaseee what did you exactly do for Solution...Thanks alot...
> 
> Open Advanced panel on bottom right and click "Reset all settings to defaults"

Hi Adrián, Advance panel in the my googles  for change de 03 settings, or in gyroflow... Thanks agan