gyroflow/gyroflow #570: Custom "save aera guide"

Mcrubens, 2023-01-14:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I think it might be useful to be able to set the safe area guide to a FOV value other than 1.0. 

Personally I don't like to use dynamic zooming as it tend to speedup the footage. So I would crop the footage with fixed FOV. I could zoom out and use the safe area guide to determine the minimal FOV. 

[image]

AdrianEddy, 2023-01-18:
Isn't that what Static zoom does?
You can also just use dynamic zoom with large zooming speed, like 30s
There's also a different zooming method: Envelope follower, which changes the zooming behavior a bit

AdrianEddy, 2023-01-18:
Latest dev build added some new features to this:
1. When using `No zooming`, safe area guide will show minimal usable frame
2. Chart will now contain red areas in places where black borders enter the video

You can see this in action here: https://discord.com/channels/797044698682228736/797045396425408522/1065105836998926376

Download: https://gyroflow.xyz/devbuild/?autodownload

closed 2023-01-18 by AdrianEddy