gyroflow/gyroflow #574: ffmpeg error::Error(22: Invalid arguement)

stargaitanis, 2023-01-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.4.2

### What operating system are you using?

Windows 10

### What GPU are you using?

980ti

### What happened?

I was able to stabilize my GP9 footage fine all this time. On 21/1/23 an error started to pop up when I tried to export the footage: 

"An error occured: ffmpeg error::Error(22: Invalid arguement), [NULL] Unable to find a suitable output format for '(file location)'.

It detects the footage and stabilizes it as usual, but when I click render I get this message and can't render anything. All render settings are the usual, but I tried playing around with it to no avail. I also tried
1. changing the footage location
2. download gyroflow again
3. tried differnt footage in case the specific file was problematic, but now I get the same error on all footage

I installed gyroflow on a Macbook and it renders it just fine. Any idea of what seems to be the problem?

AdrianEddy, 2023-01-21:
Whats your export file extension? Please also upload gyroflow.log file (Show package contents -> Contents -> MacOS)

stargaitanis, 2023-01-22:
Clicked on the red "Reset all settings to default" (bottom right), changed the settings back to my preference and all is well again. Very strange...

AdrianEddy, 2023-01-22:
I think this error is when you have wrong export file extension for the selected codec, so resetting to defaults fixed that

closed 2023-01-22 by AdrianEddy

shmelkin, 2025-06-21:
For me it was, that I had GPU encoding enabled while not having a GPU installed in my notebook