gyroflow/gyroflow #575: DJI ACTION2 GF1.4.2  horizon locking function

hzjianglf, 2023-01-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4.2

### What operating system are you using?

win10

### What happened?

Set the horizon locking function in GF, and the horizon after locking is more inclined .  I use dji action2 the newest firmware: 01. 04.04.20

[image]

[image]

AdrianEddy, 2023-01-22:
Can you share the original video file?

hzjianglf, 2023-01-24:
> Can you share the original video file?

I will upload video later。Thank you。

hzjianglf, 2023-01-25:
> Can you share the original video file?

PLS SEE THIS video @baidu netdisk .   file size is about 1.2G . Thank You.

https://pan.baidu.com/s/1ENcJKQAqqWxsImyc9gx2hQ?pwd=1234

hzjianglf, 2023-01-25:
> > Can you share the original video file?
> 
> PLS SEE THIS video @baidu netdisk . file size is about 1.2G . Thank You.
> 
> https://pan.baidu.com/s/1ENcJKQAqqWxsImyc9gx2hQ?pwd=1234

@AdrianEddy

hzjianglf, 2023-01-25:
VIDEO is uploaded 。 I Looking forward to the results of your analysis。Thank You.

jesuslg123, 2023-01-28:
Hello,

I think I facing the same issue on the latest build of Gyroflow with the same firmware. I think the horizon lock does not fail all the time, only in some part of the video.

[image]
[image]

hzjianglf, 2023-01-29:
> Hello,
> 
> I think I facing the same issue on the latest build of Gyroflow with the same firmware. I think the horizon lock does not fail all the time, only in some part of the video.
> 
> [image]
> 
> [image]

BUT when i use gopro, No similar situation will occur。

AdrianEddy, 2023-01-29:
Yes, it's a bug in DJI Action 2 firmware, I sent your file to DJI, I hope they will fix it in the next firmware

AdrianEddy, 2023-03-07:
I'm closing this, as there's nothing in Gyroflow that I can do to fix it. Hopefully it's fixed in the latest DJI Action 2 firmware.

closed 2023-03-07 by AdrianEddy