gyroflow/gyroflow #576: ffmpeg error when starting an H.264/AVC or H.265/HEVC export

loloshadow, 2023-01-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.4.2

### What operating system are you using?

Windows 10 Version 22H2 pour x64

### What GPU are you using?

NVIDIA 940MX

### What happened?

This error appears when starting an export, GPU pilot is updated. Any idea ?  

[image]

AdrianEddy, 2023-01-22:
Your GPU doesn't support accelerated rendering. Uncheck "Use GPU encoding"

closed 2023-01-22 by AdrianEddy

loloshadow, 2023-01-22:
Works well now, thanks a lot !!!