gyroflow/gyroflow #580: rotate video support

yafinewbie, 2023-01-29:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

can gyroflow have features for rotating video when recorded on portrait mode (rotated 90 deg)

AdrianEddy, 2023-01-29:
It's already there: Rotation in Video information tab

closed 2023-01-29 by AdrianEddy

yafinewbie, 2023-01-29:
it only rotate the video source but broke the aspect ratio and motion data when importing from runcam 5 orange
[image]
and export resolution is limited (can't even go 2.7k on portrait)

AdrianEddy, 2023-01-29:
What exactly do you want to do? Did you film horizontally or vertically? Do you want to export horizontal or vertical? The output resolution limitation is a limitation of H264 encoder and I can't do anything about it. You can use H265 or ProRes

What do you mean it broke the aspect ratio?

AdrianEddy, 2023-01-29:
Depending on your use case, you can also rotate the motion data, using Motion data -> Rotation -> Roll angle