gyroflow/gyroflow #581: Gyroflow 1.3.0and 1.4.2 can not render multiple videos at the same time

TierraErde, 2023-01-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.3.0 and 1.4.2

### What operating system are you using?

Mac OS 13.1

### What GPU are you using?

Radeon Pro VEGA 20

### What happened?

Gyroflow 1.3.0and 1.4.2 can not render multiple videos at the same time.
If i render one video one time,it will success，if i change video when it rendering last video,it will failed,both i can't render the new video with GPU.

[image]
[image]
[image]
[image]

[image]
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10542638/gyroflow.log)

AdrianEddy, 2023-02-10:
I can't reproduce this on M1 and M2 mac, please send me a log from latest dev build ( https://gyroflow.xyz/devbuild/?autodownload ) when it shows the error and then you close the app (the log will be complete after you close the app)

AdrianEddy, 2023-03-07:
Do you still have this issue with latest dev build? https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2023-05-29:
I don't think there's anything I can do about it on my side, I think it's a limitation of the GPU encoders in Intel Mac devices

closed 2023-05-29 by AdrianEddy