gyroflow/gyroflow #582: VQF works incorrectly if log length does not match video length

VladimirP1, 2023-02-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.4.2 gh1078

### What operating system are you using?

windows 11

### What happened?

Steps to reproduce:
1. Find a video with a separate log (gcsv, blackbox, runcam, does not matter)
2. Check that the video stabilizes correctly with VQF
3. Truncate the log to half of its length
4. Try to stabilize the video again

Notes:
It seems that `sample_time` [here](https://github.com/gyroflow/gyroflow/blob/master/src/core/imu_integration/mod.rs#L107) gets calculated based on video length, not log length, so the motion gets multiplied by (video duration)/(log duration).

VladimirP1, 2023-02-01:
Probably it is enough just to properly calculate log length by taking the last minus the first timestamp in IMUData. 
Or maybe interpolation is needed? For the case of non uniform samples?

VladimirP1, 2023-02-01:
https://github.com/gyroflow/gyroflow/pull/583

AdrianEddy, 2023-02-01:
Fixed in 3eb4e0facb3c43812a5fcf466c89dd43ae8573f3, thanks!

closed 2023-02-01 by AdrianEddy