gyroflow/gyroflow #585: The preview issue

ghostkcleung, 2023-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.4.2(gh972)

### What operating system are you using?

Intel® Core™ i7-7700HQ

### What GPU are you using?

NVIDIA GeForce GTX 1050 Ti 

### What happened?

Can not play preview.

https://drive.google.com/file/d/1KyAeRF7j-1aIJncEsYXSZKaSNPe4U_sF/view?usp=sharing

AdrianEddy, 2023-02-11:
1. Install latest driver from https://www.nvidia.com/download/driverResults.aspx/199654/en-us/
2. Does the preview work if you toggle stabilisation off? (S key)
3. Does anything change if you select different Preview pipeline under Advanced?

AdrianEddy, 2023-03-07:
Do you still have this issue?

ghostkcleung, 2023-03-07:
> Do you still have this issue with latest dev build? https://gyroflow.xyz/devbuild/?autodownload

It's OK with the latest nightly build. Thank you.

closed 2023-03-07 by AdrianEddy