gyroflow/gyroflow #587: ffmpeg error wont export Hero 10 bones 5k 4:3 files

DivingFalconFPV, 2023-02-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4.2

### What operating system are you using?

wins 10

### What GPU are you using?

vega 64

### What happened?

Wont export stabilize the video files. 
[image]

DivingFalconFPV, 2023-02-10:
if it matter also reelsteady wont either. That gave me a error too. files play though

AdrianEddy, 2023-02-10:
Do you have latest GPU driver? What's the exact model of your GPU?

DivingFalconFPV, 2023-02-10:
I lowered it to 4k it worked.
I am not even sure if latest drivers. Normally it tells me on catalyst app
to update.
It's Radeon Vega 64
But i had to stop the render cause computer was over heating (close to it
dying)  looked my water cooling system clogged. Have to fix it

DivingFalconFPV, 2023-02-10:
Also believe I turned off using gpu

AdrianEddy, 2023-02-11:
Install latest driver from here https://www.amd.com/en/support/graphics/radeon-rx-vega-series/radeon-rx-vega-series/radeon-rx-vega-64

I'm not sure which encoding modes this GPU supports, but in any case try lower resolution and H264 instead of H265.

closed 2023-02-11 by AdrianEddy