gyroflow/gyroflow #589: Custom encoder options

victorlysak, 2023-02-14:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hi.

This is either a wish or a question :)

My system is Manjaro Linux (archlinux) with xfce desktop.

Gyroflow works very well.

However.
I am converting "4k50pfs" video from "x265 aac" to DNXHR444 format, and in the custom encoder options I'm giving
-c:a pcm_s16le
so I can have the uncompressed sound, but the final file still has aac sound.

Does that field refers to only encoders or is that the normal command line addition to the ffmpeg?

And would it be possible to add similar field so it would contain custom params for the ffmpeg?

The reason I do this is because I need to create intermed. for further processing in Davinci Resolve.
The problem with Resolve is that it will not accept aac sound, and that means that I need to have the videos with pcm_s16le (or something like that).

Thank you.

AdrianEddy, 2023-02-14:
These are only video encoder options, the audio is hardcoded to always be AAC.
It's not easy to provide this, because Gyroflow doesn't call ffmpeg command line, so all custom options would have to be reimplemented manually.

What do you mean by Resolve not accepting AAC? I always use Resolve with AAC audio without issues

victorlysak, 2023-02-15:
Hi.

quote:
     What do you mean by Resolve not accepting AAC? I always use Resolve with AAC audio without issues

Yes, but not under Linux, as a free version, where you don't have aac, h264, h265, and many others :)

Thanks for your reply.

AdrianEddy, 2023-05-29:
Done in 0e4b97826f2e430ef2323a7a15f468749b5c9d88

closed 2023-05-29 by AdrianEddy