gyroflow/gyroflow #590: the gyro magnitude calculation in get_adaptive_gain() of complementary_v2 is incorrect

ClarkWang1214, 2023-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.4.2

### What operating system are you using?

Windows 11

### What GPU are you using?

RTX 3060 Laptop GPU

### What happened?

It seems that the calculation about w_mag[(here)](https://github.com/gyroflow/gyroflow/blob/master/src/core/imu_integration/complementary_v2.rs#L414) is incorrect. It's maybe just a little glitch.  I think the correct one is "let w_mag = (self.w_prev.0*self.w_prev.0 + self.w_prev.1*self.w_prev.1 + self.w_prev.2*self.w_prev.2).sqrt();"

AdrianEddy, 2023-02-16:
Good catch, updated now.

I'm curious, how did you find it?

closed 2023-02-16 by AdrianEddy

ClarkWang1214, 2023-02-17:
> Good catch, updated now.
> 
> I'm curious, how did you find it?

I have compiling and ran the Gyroflow on my laptop.  Meanwhile, I have read all of the source code for Gyroflow and I'm trying my best to understand what each module means.  

Because i want to transplant the Gyroflow so as to implement a real-time version Video Stabilization tool based on EIS using C launguage.    

Could you give me some suggestion?  

THX.

AdrianEddy, 2023-02-17:
I'm currently in the process of writing comprehensive documentation about Gyroflow including technical stuff. It's not done yet, but take a look at the "Technical details" section here: https://gyroflow.gitbook.io/gyroflow-docs/technical-details/used-technologies

also check out https://gyroflow.gitbook.io/gyroflow-docs/advanced-usage/live-feed-stabilization

If the live feed stabilization would be added to core, then you could just compile gyroflow_core with a C interface (like FCPX plugin does: [header](https://github.com/latenitefilms/GyroflowToolbox/blob/main/Source/Frameworks/gyroflow/inc/gyroflow.h) + [impl](https://github.com/latenitefilms/GyroflowToolbox/blob/main/Source/Frameworks/gyroflow/src/lib.rs#L37-L38) + [usage](https://github.com/latenitefilms/GyroflowToolbox/blob/main/Source/Gyroflow/Plugin/GyroflowPlugIn.m#L1041) ) instead of rewriting everything to C (which would be a huge task)

ClarkWang1214, 2023-02-17:
> 

Thanks for your reply. I will read them.

- Besides, when I calibrated my own camera without any lens presets, the size of my printed chessboard is 11 x 8, so as to the calibration is falure( **0 good frames and the Reprojection error is 999.0**).

[image]
[image]

- After i read the source code [(here)](https://github.com/gyroflow/gyroflow/blob/master/src/core/calibration/mod.rs#L75), I found that the chessboard used in the Lens calibrator is fixed (14 x 8). 

- Then, I added two labels in the **LensCalibrate.qml**, It is convenient for me to insert the number of rows and columns according to my own chessboard. 

[image]

- Meanwhile, I changed the **start_autocalibrate()** function and other related files. Finally, it's successful for my own chessboard.

[image]

The above are some attempts I made according to my own situation. If possible, I hope to provide some improvements, Thx.

ClarkWang1214, 2023-02-17:
> I'm currently in the process of writing comprehensive documentation about Gyroflow including technical stuff. It's not done yet, but take a look at the "Technical details" section here: [...]

Thanks for your reply. I will read them.

**Besides**, when I calibrated my own camera without any lens presets, the calibration failed due to the size of the chessboard i printed at 11 x 8 (**0 good frames and the Reprojection error is 999.0**). 

[image]
[image]

- After I read the source code [(here)](https://github.com/gyroflow/gyroflow/blob/master/src/core/calibration/mod.rs#L75), I found that the chessboard used in the Lens calibrator is fixed (14 x 8). 

- Then, I added two labels in the **LensCalibrate.qml**, It is convenient for me to insert the number of rows and columns according to my own chessboard. 

[image]

- **Meanwhile, I changed the **start_autocalibrate()** function and other related files. Finally, it's successful for my own chessboard.**

[image]

**The above are some attempts I made according to my own situation. If possible, I hope to provide some improvements, Thx.**