gyroflow/gyroflow #591: Green screen preview

Mielotxin, 2023-02-16:
Well, it was working great, but after the update I only get green screens on my gopro 8 and 10 previews. The BMPCC and the gopro11 are working great tho...
Gyro reading, sound and export are working fine. Did I mess any setting?
TIA

[image]

AdrianEddy, 2023-02-16:
Did you try latest dev build? https://gyroflow.xyz/devbuild/?autodownload

Mielotxin, 2023-02-17:
Not yet, but I managed to make it work in a strange way. I first load a GP11 clip and then the GP8 one loads perfectly.
It seems to be some random bug? Thank you Eddy!

AdrianEddy, 2023-02-17:
Please test the latest dev build from the link I provided, I think it should be fixed there

AdrianEddy, 2023-03-07:
Do you still have this issue?

Mielotxin, 2023-03-08:
Yes! I'm sorry for the late reply...
Thank you for your time Eddy, and the v1.5 it's looking solid too!

AdrianEddy, 2023-03-08:
Yes as in "yes it's fixed" or "yes, I still have the green screen issue"? :)

Mielotxin, 2023-03-08:
Yes, it's fixed.

closed 2023-03-08 by AdrianEddy