gyroflow/gyroflow #594: Background mode blur + feather options

evoth, 2023-03-03:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hi, although I understand it may be compute-intensive, I would love to have an option to blur/feather the background (see [this message](https://discord.com/channels/797044698682228736/797044698682228743/966047929762721822) I found on the Discord for an example of what I'm looking for). It could be a general option for the background, but I'm most interested in the ability to blur the "Repeat edge pixels" option to reduce the distracting look it currently produces.

I hope this is not too hard to add, but I understand that it's probably low priority. Thank you for working on this wonderful tool!

AdrianEddy, 2023-03-03:
Blur of any kind is hard, because it needs to be done in 2 passes, which current pipeline doesn't support

evoth, 2023-03-03:
In that case, is there a way to export a separate background mask for automated processing in an external tool? For example, in this "mask" video, the background pixels would be white and video pixels would be black (or vice versa).

AdrianEddy, 2023-03-03:
you can export transparent video, and then use alpha mask in any video editor. Simply put "transparent" in the Background color field and use ProRes 4444

evoth, 2023-03-03:
> Simply put "transparent" in the Background color field and use ProRes 4444

Perfect, I didn't know that was possible!