gyroflow/gyroflow #595: nightly AppImage broken

dotSynergy, 2023-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

gh1124

### What operating system are you using?

Pop!_OS 22.04 LTS

### What GPU are you using?

GTX 1070

### What happened?

AppImage crashes on startup, this has been happening for a couple weeks on nightlies (tried as far back as a build on the 14th of feb). Release AppImage works fine.
It is launched through AppImageLauncher but the behavior is the same by launching it regularly. Plain old executable is 100% fine.

attached crashdump (zipped)
[ea5dd34c-b1fe-48c8-f258bda1-a470cd0a.zip](https://github.com/gyroflow/gyroflow/files/10913907/ea5dd34c-b1fe-48c8-f258bda1-a470cd0a.zip)

### Relevant log output

```shell
Mar  7 21:56:48 pop-os nautilus[23679]: 20:56:48 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
Mar  7 21:56:48 pop-os nautilus[23679]: 20:56:48 [DEBUG] (1) gyroflow: Gyroflow 1.4.2 (gh1124)
Mar  7 21:56:48 pop-os nautilus[23679]: 20:56:48 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.0.139") ["NVIDIA GeForce GTX 1070"]
Mar  7 21:56:48 pop-os nautilus[23679]: 20:56:48 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation NVIDIA GeForce GTX 1070
Mar  7 21:56:48 pop-os nautilus[23679]: 20:56:48 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
Mar  7 21:56:48 pop-os nautilus[23679]: 20:56:48 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
Mar  7 21:56:48 pop-os nautilus[23679]: 20:56:48 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
Mar  7 21:56:48 pop-os nautilus[23679]: 20:56:48 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce gtx 1070
Mar  7 21:56:48 pop-os gnome-shell[2327]: Can't update stage views actor <unnamed>[<MetaWindowGroup>:0x5640f7544370] is on because it needs an allocation.
Mar  7 21:56:48 pop-os gnome-shell[2327]: Can't update stage views actor <unnamed>[<MetaWindowActorX11>:0x5640fb051700] is on because it needs an allocation.
Mar  7 21:56:48 pop-os gnome-shell[2327]: Can't update stage views actor <unnamed>[<MetaSurfaceActorX11>:0x5640f7173f60] is on because it needs an allocation.
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce GTX 1070: 3.0", "[wgpu] NVIDIA GeForce GTX 1070 (Vulkan)", "[wgpu] llvmpipe (LLVM 15.0.6, 256 bits) (Vulkan)", "[wgpu] NVIDIA GeForce GTX 1070/PCIe/SSE2 (Gl)"]
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [INFO] Lens profiles directory: "/home/dotsynergy/Downloads/Gyroflow/camera_presets"
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: gopro-hero6black-linear-2704x2028@25000-no-eis, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/GoPro/GoPro_HERO6 Black_Linear_NO-EIS_4by3.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/GoPro/GoPro_HERO6 Black_Linear_ _2.7k_4by3_2704x2028-25.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: gopro-hero6black-linear-2704x2028@59940-no-eis, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/GoPro/GoPro_HERO6 Black_Linear_NO-EIS_2.7k_4by3_2704x2028-59.94fps.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/GoPro/GoPro_HERO6 Black_Linear_NO-EIS_4by3.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: gopro-hero6black-linear-1920x1080@29970-no-eis, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/GoPro/GoPro_HERO6 Black_Linear_NO-EIS_1080p_16by9_1920x1080-29.97fps.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/GoPro/GoPro_HERO6 Black_Linear_NO-EIS_16by9.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: sony-ilme-fx6v-14.00mm-3840x2160@50000, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_Sony GM 14.00mm__4k_16by9_3840x2160-50.00fps.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_14.00mm__4k_16by9_3840x2160-50.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: sony-ilme-fx6v-35.00mm-3840x2160@50000, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_35.00mm__4k_16by9_3840x2160-50.00fps.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_Sony GM 35.00mm__4k_16by9_3840x2160-50.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: sony-ilme-fx6v-35.00mm-4096x2160@50000, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_Sony GM 35.00mm__C4k_1.90by1_4096x2160-50.00fps.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_35.00mm__C4k_1.90by1_4096x2160-50.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: sony-ilme-fx6v-20.00mm-3840x2160@50000, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_Sony GM 20.00mm__4k_16by9_3840x2160-50.00fps.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_20.00mm__4k_16by9_3840x2160-50.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: sony-ilme-fx6v-20.00mm-4096x2160@50000, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_Sony GM 20.00mm__C4k_1.90by1_4096x2160-50.00fps.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_20.00mm__C4k_1.90by1_4096x2160-50.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [WARN] Lens profile already present: sony-ilme-fx6v-14.00mm-4096x2160@50000, filename: /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_Sony GM 14.00mm__C4k_1.90by1_4096x2160-50.00fps.json from /home/dotsynergy/Downloads/Gyroflow/camera_presets/Sony/Sony_FX6_14.00mm__C4k_1.90by1_4096x2160-50.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [INFO] Loaded 6694 lens profiles in 193.543ms
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [ERROR] Duplicated lens profile! /home/dotsynergy/Downloads/Gyroflow/camera_presets/Panasonic/Panasonic_GH5_Laowa 7.5mm F 2 MFT__4k_16by9_3840x2160-29.97fps.json vs /home/dotsynergy/Downloads/Gyroflow/camera_presets/Panasonic/Panasonic_GH5_Laowa 7.5mm f2.0 MFT__4k_16by9_3840x2160-29.97fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [ERROR] Duplicated lens profile! /home/dotsynergy/Downloads/Gyroflow/camera_presets/RED/RED_KOMODO_Laowa 9mm_30 fps prores_4k_16by9_3840x2160-29.97fps.json vs /home/dotsynergy/Downloads/Gyroflow/camera_presets/RED/RED_KOMODO_laowa 9 mm_30 fps prores_4k_16by9_3840x2160-29.97fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [ERROR] Duplicated lens profile! /home/dotsynergy/Downloads/Gyroflow/camera_presets/Panasonic/Panasonic_GH5_Laowa 7.5mm f2.0__1080p_16by9_1920x1080-25.00fps.json vs /home/dotsynergy/Downloads/Gyroflow/camera_presets/Panasonic/Panasonic_GH5_Lawoa 7,5MM 2.0__1080p_16by9_1920x1080-25.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [ERROR] Duplicated lens profile! /home/dotsynergy/Downloads/Gyroflow/camera_presets/RED/RED_Raven_Cann EF 50mm L f1.2__C4k_2.13by1_4608x2160-25.00fps.json vs /home/dotsynergy/Downloads/Gyroflow/camera_presets/RED/RED_Raven_Canon EF 50mm L f1.2__C4k_2.13by1_4608x2160-25.00fps.json
Mar  7 21:56:49 pop-os nautilus[23679]: 20:56:49 [ERROR] Duplicated lens profile! /home/dotsynergy/Downloads/Gyroflow/camera_presets/Panasonic/panasonic_GH5_Laowa_7.5mm_4k_16by9_3840x2160-50.00fps - 2.json vs /home/dotsynergy/Downloads/Gyroflow/camera_presets/Panasonic/panasonic_GH5_Laowa_stab on 7.5mm_4k_16by9_3840x2160-50.00fps.json
[... cut ...]
Mar  7 21:57:01 pop-os kernel: [ 9127.378560] gyroflow[23754]: segfault at 20c41a0b8 ip 00007f12c401008d sp 00007ffe0991ddc0 error 4 in libxkbcommon-x11.so.0.0.0[7f12c400f000+4000] likely on CPU 7 (core 7, socket 0)
Mar  7 21:57:01 pop-os kernel: [ 9127.378573] Code: 0f 84 b9 00 00 00 48 83 c3 01 48 39 5c 24 08 0f 84 33 02 00 00 48 8b 04 24 48 21 d8 74 e8 49 8b 14 24 48 8d 0c 85 00 00 00 00 <44> 8b 3c 82 45 85 ff 75 aa 45 8b 7c 24 18 41 8b 44 24 1c 41 8d 57
Mar  7 21:57:01 pop-os systemd[1]: tmp-.mount_GyroflnYPO2C.mount: Deactivated successfully.
```

AdrianEddy, 2023-03-07:
Great, just as I released v1.5.0 🐵
I won't be able to look into it until next week though, but thanks for reporting!

froqstar, 2023-03-08:
1.5.0 AppImage crashes here as well:

```
➜  Downloads ./Gyroflow-linux64.AppImage 
MDK 0.20.0 (git 3dd65ad) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.0 (++20230306093126+ae37edf1486d-1~exp1~20230306213138.48)GCC4.2.1; libc++160000; 01:35:31 Mar  7 2023
10:52:45 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
10:52:45 [DEBUG] (1) gyroflow: Gyroflow 1.5.0 (gh1128)
mdk plugin 'mdk-braw': 0x560c476dff90 from dir: /tmp/.mount_GyroflMXu1Um//lib/
mdk plugin 'mdk-r3d': 0x560c476e1dd0 from dir: /tmp/.mount_GyroflMXu1Um//lib/
10:52:45 [DEBUG] (4) gyroflow::util: Minidump uploaded: OK
10:52:45 [ERROR] thread 'main' panicked at 'called `Option::unwrap()` on a `None` value': /root/.cargo/registry/src/github.com-1ecc6299db9ec823/khronos-egl-4.1.0/src/lib.rs:1518
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: main
  14: __libc_start_call_main
  15: __libc_start_main@GLIBC_2.2.5
  16: <unknown>

10:52:45 [ERROR] Failed to initialize wgpu called `Option::unwrap()` on a `None` value
/usr/lib/gio/modules/libdconfsettings.so: wrong ELF class: ELFCLASS32
Failed to load module: /usr/lib/gio/modules/libdconfsettings.so

(gyroflow:119036): GLib-GIO-ERROR **: 11:52:46.404: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'
Crashdump written to /home/froqstar/Downloads/90b3cf24-e2ad-4186-be4cfcbc-bf4a8a48.dmp
[1]    119036 trace trap (core dumped)  ./Gyroflow-linux64.AppImage
```

[Zipped crashdump](https://github.com/gyroflow/gyroflow/files/10919843/dump.zip)

AdrianEddy, 2023-03-14:
AppImage should be fixed in latest dev build, please check https://gyroflow.xyz/devbuild/?autodownload

froqstar, 2023-03-14:
Works for me, thanks a lot :)

closed 2023-03-14 by AdrianEddy