gyroflow/gyroflow #596: App Crash: Cannot load nvEncodeAPI64.dll

obreo, 2023-03-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

Windows 11 22H2

### What GPU are you using?

NVIDIA GeForce MX150

### What happened?

Whenever I try to export the footage, I get an error saying: Cannot load nvEncodeAPI64.dll, and it crashes. 

My NVIDIA driver's version is 528.49

I also tried to change the settings to use NVIDIA under Vulkan, DirectX12, and OpenCL but I keep getting the same error.

I have googled the error and it seems h264_nvenc is not supported by my NVIDIA, although I used Nvidia driver while using Davinci Resolve and i didn't get such error when exporting. Here are links:
[https://forums.developer.nvidia.com/t/mx150-seems-not-to-support-nvdec-nvenc-in-windows-10/77628/3](url)
[https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new](url)

I tried switching to Intel's graphic and it exports but takes long time and the output is damaged (Only sound but green screen).

Any suggestions to bypass this issue?

Thanks,

### Relevant log output

```shell
05:26:37 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
05:26:37 [DEBUG] (1) gyroflow: Gyroflow 1.5.0
05:26:39 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.0.147") ["NVIDIA GeForce MX150"]
05:26:39 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL") Ok("OpenCL 2.1") ["Intel(R) UHD Graphics 620"]
05:26:39 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce MX150
05:26:39 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
05:26:39 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
05:26:39 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
05:26:39 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce mx150
05:26:40 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce MX150: 3.0", "[OpenCL] Intel(R) OpenCL Intel(R) UHD Graphics 620: 2.1", "[wgpu] NVIDIA GeForce MX150 (Vulkan)", "[wgpu] Intel(R) UHD Graphics 620 (Vulkan)", "[wgpu] NVIDIA GeForce MX150 (Dx12)", "[wgpu] Intel(R) UHD Graphics 620 (Dx12)"]
05:26:41 [INFO] Lens profiles directory: "\\\\?\\C:\\Program Files (x86)\\Gyroflow-windows64\\camera_presets"
05:26:42 [INFO] CameraIdentifier {
    brand: "GoPro",
    model: "HERO10 Black",
    lens_model: "",
    lens_info: "Wide",
    focal_length: None,
    camera_setting: "",
    fps: 29970,
    video_width: 1920,
    video_height: 1080,
    additional: "HS High",
    identifier: "gopro-hero10black-wide-1920x1080@29970-hshigh",
}
05:26:42 [DEBUG] (5) telemetry_parser::util: Reading duration: (Some(4.994934523140551), Some(4.994934523140551), None)
05:26:42 [INFO] Loaded 6390 lens profiles in 946.199ms
05:26:42 [WARN] IMU duration 223022.79467459602 is different than video duration (223023)
05:26:42 [WARN] IMU duration 223053.7958226666 is different than video duration (223022.79467459602)
05:26:42 [INFO] No gravity vectors - using accelerometer
[... 90 lines cut ...]
[hevc @ 0000016766685E00] Failed setup for format dxva2_vld: hwaccel initialisation returned error.
05:27:15 [ERROR] [hevc @ 0000016766685E00] Failed setup for format dxva2_vld: hwaccel initialisation returned error.
05:27:15 [DEBUG] (2) gyroflow_core::stabilization: Setting device 4
05:27:15 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU adapter (2): AdapterInfo { name: "NVIDIA GeForce MX150", vendor: 4318, device: 7440, device_type: DiscreteGpu, driver: "", driver_info: "", backend: Dx12 }
05:27:15 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU initializing adapter #2
05:27:17 [INFO] Initialized wgpu for (1920, 1080, 1920) -> (1920, 1080, 1920) | key: 554466488opencv_fisheye80(1920, 1080)(1920, 1080)
05:27:17 [DEBUG] (2) gyroflow_core::stabilization: Setting device 4
05:27:17 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU adapter (2): AdapterInfo { name: "NVIDIA GeForce MX150", vendor: 4318, device: 7440, device_type: DiscreteGpu, driver: "", driver_info: "", backend: Dx12 }
05:27:17 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU initializing adapter #2
05:27:18 [INFO] Initialized wgpu for (960, 540, 960) -> (960, 540, 960) | key: 4104633768opencv_fisheye80(960, 540)(960, 540)
05:27:18 [DEBUG] (2) gyroflow_core::stabilization: Setting device 4
05:27:18 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU adapter (2): AdapterInfo { name: "NVIDIA GeForce MX150", vendor: 4318, device: 7440, device_type: DiscreteGpu, driver: "", driver_info: "", backend: Dx12 }
05:27:18 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU initializing adapter #2
05:27:19 [INFO] Initialized wgpu for (960, 540, 960) -> (960, 540, 960) | key: 4104633768opencv_fisheye80(960, 540)(960, 540)
05:27:19 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: None
05:27:19 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
05:27:19 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: JPEG
05:27:19 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_CUDA)
05:27:19 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x167908097c0
[h264_nvenc @ 0000016790F557C0] Cannot load nvEncodeAPI64.dll
05:27:19 [ERROR] [h264_nvenc @ 0000016790F557C0] Cannot load nvEncodeAPI64.dll
[h264_nvenc @ 0000016790F557C0] nvenc_load_functions error05:27:19 [ERROR] [h264_nvenc @ 0000016790F557C0] nvenc_load_functions error
The minimum required Nvidia driver for nvenc is 522.25 or newer
05:27:19 [ERROR] The minimum required Nvidia driver for nvenc is 522.25 or newer
```

AdrianEddy, 2023-03-08:
GeForce MX150 doesn't support any hardware encoding so you have to use Intel or CPU
https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new

closed 2023-03-08 by AdrianEddy

fhajji, 2023-03-21:
This seems to be the case with the MX250 as well:

```
23:33:23 [INFO] [MDK] MDK 0.20.0 (git 3dd65ad) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1934; MSVCRT14.34.31942.0/msvcp140.dll; 01:43:57 Mar  7 2023
23:33:23 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
23:33:23 [DEBUG] (1) gyroflow: Gyroflow 1.5.0
23:33:23 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
23:33:23 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
23:33:23 [DEBUG] (1) mdk: [MDK] user module: D:\Downloads\Software\Blackmagic_Design\Gyroflow-windows64-1.5.0\Gyroflow.exe
23:33:23 [DEBUG] (1) mdk: [MDK] user module: D:\Downloads\Software\Blackmagic_Design\Gyroflow-windows64-1.5.0\Gyroflow.exe
23:33:23 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
23:33:27 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.1.68") ["NVIDIA GeForce MX250"]
23:33:27 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") Ok("OpenCL 3.0") ["Intel(R) UHD Graphics 620"]
23:33:27 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce MX250
23:33:27 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
23:33:27 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
23:33:27 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
23:33:27 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce mx250
23:33:31 [INFO] Latest version: v1.5.0, current version: 1.5.0
23:33:31 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce MX250: 3.0", "[OpenCL] Intel(R) OpenCL HD Graphics Intel(R) UHD Graphics 620: 3.0", "[wgpu] NVIDIA GeForce MX250 (Vulkan)", "[wgpu] Intel(R) UHD Graphics 620 (Vulkan)", "[wgpu] NVIDIA GeForce MX250 (Dx12)", "[wgpu] Intel(R) UHD Graphics 620 (Dx12)"]
23:33:31 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia cuda nvidia geforce mx250: 3.0
23:33:31 [INFO] CameraIdentifier {
    brand: "GoPro",
    model: "HERO7 Black",
    lens_model: "",
    lens_info: "Wide",
    focal_length: None,
    camera_setting: "",
    fps: 59940,
    video_width: 3840,
    video_height: 2160,
    additional: "NO-EIS",
    identifier: "gopro-hero7black-wide-3840x2160@59940-no-eis",
}
23:33:31 [DEBUG] (2) telemetry_parser::util: Reading duration: (Some(5.044519901905301), Some(5.044519901905301), None)
23:33:32 [WARN] IMU duration 187186.9993839437 is different than video duration (187320)
23:33:32 [INFO] Lens profiles directory: "\\\\?\\D:\\Downloads\\Software\\Blackmagic_Design\\Gyroflow-windows64-1.5.0\\camera_presets"
23:33:42 [INFO] Loaded 6390 lens profiles in 9559.466ms
23:33:43 [WARN] [mov,mp4,m4a,3gp,3g2,mj2 @ 00000236E0BE7300] All samples in data stream index:id [4:5] have zero duration, stream set to be discarded by default. Override using AVStream->discard or -discard for ffmpeg command.
23:33:56 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
23:33:56 [WARN] [mov,mp4,m4a,3gp,3g2,mj2 @ 00000236E50311C0] All samples in data stream index:id [4:5] have zero duration, stream set to be discarded by default. Override using AVStream->discard or -discard for ffmpeg command.
23:33:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
23:33:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_DXVA2, device: None
23:33:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)
23:33:56 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("dxva2")
23:33:56 [DEBUG] (2) gyroflow::rendering: Possible encoders with Nvidia: [("hevc_nvenc", true), ("hevc_mf", true), ("libx265", false)]
23:33:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
23:33:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: HW formats: [CUDA]
23:33:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: SW formats: [NV12, YUV420P, YUVA420P, YUV444P, P010LE, P016LE, YUV444P16LE, BGRZ, RGBZ]
23:33:56 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_nvenc"), processing_order: PreConversion
23:33:56 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 187320, render_frame_count: 11228
23:33:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, NV12, P010LE, YUV444P, P016LE, YUV444P16LE, BGRZ, BGRA, RGBZ, RGBA, X2RGB10LE, X2BGR10LE, GBRP, GBRP16LE, CUDA, D3D11]
23:33:57 [ERROR] [hevc @ 00000236A88C03C0] Failed setup for format dxva2_vld: hwaccel initialisation returned error.
[... cut ...]
23:33:58 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x236b04e4140
23:33:58 [ERROR] [hevc_nvenc @ 00000236A88C43C0] Cannot load nvEncodeAPI64.dll
23:33:58 [ERROR] [hevc_nvenc @ 00000236A88C43C0] nvenc_load_functions error
23:33:58 [ERROR] The minimum required Nvidia driver for nvenc is 522.25 or newer
```

The driver version is 531.29 03/14/2023.

I've set Windows to use the NVIDIA GeForce MX250 card in Windows Systems / Graphics settings / Graphics performance to ensure Gyroflow.exe doesn't use the Intel UHD Graphics 620. Still, no luck there.

AdrianEddy, 2023-03-22:
You need to set "Device for rendering" to Intel, in Export settings

"Device for video processing" can be NVIDIA, and [wgpu] is preferred on windows

fhajji, 2023-03-22:
Yes, that was it! Thank you so much. It works now as expected.