gyroflow/gyroflow #600: GPU accelerated encoder doesnt support this pixel format (YUV420P)

neurot1cal, 2023-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

Windows 10 19045.2673

### What GPU are you using?

Intel UHD 630 (i7 8700K)

### What happened?

Expecting to be able to use GPU acceleration to export, but being forced to use 'Render using CPU'. Latest Intel drivers installed (31.0.101.2115).

AdrianEddy, 2023-03-13:
There's nothing that Gyroflow can do about it, it's up to the hardware

closed 2023-03-13 by AdrianEddy

neurot1cal, 2023-03-13:
> There's nothing that Gyroflow can do about it, it's up to the hardware

GoPro Player w/ Reelsteady has no issues. GPU usage 100% and processes very quickly. :(

AdrianEddy, 2023-03-13:
GoPro player doesn't use ffmpeg, and the decoders and encoders are different. The pixel format conversion in Gyroflow is done on the CPU, and in the GoPro Player is done on the GPU. It's a challenging task that requires to rewrite a lot of code. It will eventually be done, but it will take a lot of time and effort. It's related to #555 #565 and #428

For the time being, Gyroflow uses the CPU to convert the pixel format.

neurot1cal, 2023-03-13:
Thanks for the clarity. Is there a different pixel format I can use on the GoPro to work around this?

AdrianEddy, 2023-03-13:
Unfortunately not in GoPro. 
You can change your GPU to support more pixel formats and NVIDIA has most supported formats.