gyroflow/gyroflow #603: gyroflow::rendering: Failed to create specified HW device: AV_HWDEVICE_TYPE_QSV

ojoojoo, 2023-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5

### What operating system are you using?

Windows 11

### What GPU are you using?

NVIDIA Quadro K4200

### What happened?

[image]
[gyroflow.zip](https://github.com/gyroflow/gyroflow/files/10979804/gyroflow.zip)

AdrianEddy, 2023-03-15:
Use latest dev build and select "Device for rendering" in Export settings -> Advanced to the Quadro card.
Also make sure you have latest NVIDIA driver installed
https://gyroflow.xyz/devbuild/?autodownload

closed 2023-03-15 by AdrianEddy

ojoojoo, 2023-03-15:
Yes, now it's working with the nightly, with version 1.5 the application crashes en quits. In the export settings I already selected the Quadro.