gyroflow/gyroflow #605: Gyroflow 1.5.0 crashes with multiple render threads

3dkopfkino, 2023-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.4.2 / 1.5.0 / 1.5.0 (gh1142)

### What operating system are you using?

Windows 10 22H2

### What GPU are you using?

Nvidia RTX 3050

### What happened?

-  Add multiple videos to queue
-  Increase number of render threads. 3 works fine, as soon as I select 4, the app crashes

Also, when starting each render process individually (right click on video in queue / render now), gyroflow does ignore the max render threads setting.
[crashdump_7173bcb0-06fd-4ab8-a03b-ff6e39edb9e1.zip](https://github.com/gyroflow/gyroflow/files/10995984/crashdump_7173bcb0-06fd-4ab8-a03b-ff6e39edb9e1.zip)

### Relevant log output

```shell
21:21:15 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
21:21:15 [DEBUG] (3) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
21:21:15 [DEBUG] (4) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
21:21:16 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
21:21:16 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_DXVA2, device: None
21:21:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
21:21:16 [DEBUG] (4) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
21:21:16 [DEBUG] (3) gyroflow::rendering::ffmpeg_processor: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)
21:21:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)
21:21:16 [DEBUG] (4) gyroflow::rendering::ffmpeg_processor: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)
21:21:16 [DEBUG] (3) gyroflow::rendering: proc.gpu_device: Some("dxva2")
21:21:16 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("dxva2")
21:21:16 [DEBUG] (3) gyroflow::rendering: Possible encoders with Nvidia: [("hevc_nvenc", true), ("hevc_mf", true), ("libx265", false)]
21:21:16 [DEBUG] (4) gyroflow::rendering: proc.gpu_device: Some("dxva2")
21:21:16 [DEBUG] (2) gyroflow::rendering: Possible encoders with Nvidia: [("hevc_nvenc", true), ("hevc_mf", true), ("libx265", false)]
21:21:16 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
21:21:16 [DEBUG] (4) gyroflow::rendering: Possible encoders with Nvidia: [("hevc_nvenc", true), ("hevc_mf", true), ("libx265", false)]
21:21:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
21:21:16 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: HW formats: [CUDA]
21:21:16 [DEBUG] (4) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
21:21:16 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: SW formats: [NV12, YUV420P, YUVA420P, YUV444P, P010LE, P016LE, YUV444P16LE, BGRZ, RGBZ]
21:21:16 [DEBUG] (3) gyroflow::rendering: video_codec: Some("hevc_nvenc"), processing_order: PostConversion
21:21:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: HW formats: [CUDA]
21:21:16 [DEBUG] (3) gyroflow::rendering: start_us: 0, render_duration: 116616.5, render_frame_count: 3495
21:21:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: SW formats: [NV12, YUV420P, YUVA420P, YUV444P, P010LE, P016LE, YUV444P16LE, BGRZ, RGBZ]
21:21:16 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_nvenc"), processing_order: PostConversion
21:21:16 [DEBUG] (4) gyroflow::rendering::ffmpeg_hw: HW formats: [CUDA]
21:21:16 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 113613.5, render_frame_count: 3405
21:21:16 [DEBUG] (4) gyroflow::rendering::ffmpeg_hw: SW formats: [NV12, YUV420P, YUVA420P, YUV444P, P010LE, P016LE, YUV444P16LE, BGRZ, RGBZ]
21:21:16 [DEBUG] (4) gyroflow::rendering: video_codec: Some("hevc_nvenc"), processing_order: PostConversion
21:21:16 [DEBUG] (4) gyroflow::rendering: start_us: 0, render_duration: 157657.5, render_frame_count: 4725
21:21:16 [DEBUG] (3) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, NV12, P010LE, YUV444P, P016LE, YUV444P16LE, BGRZ, BGRA, RGBZ, RGBA, X2RGB10LE, X2BGR10LE, GBRP, GBRP16LE, CUDA, D3D11]
21:21:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, NV12, P010LE, YUV444P, P016LE, YUV444P16LE, BGRZ, BGRA, RGBZ, RGBA, X2RGB10LE, X2BGR10LE, GBRP, GBRP16LE, CUDA, D3D11]
21:21:16 [DEBUG] (4) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, NV12, P010LE, YUV444P, P016LE, YUV444P16LE, BGRZ, BGRA, RGBZ, RGBA, X2RGB10LE, X2BGR10LE, GBRP, GBRP16LE, CUDA, D3D11]
21:21:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12])
21:21:16 [DEBUG] (4) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12])
21:21:16 [DEBUG] (3) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12])
21:21:16 [DEBUG] (2) gyroflow_core::stabilization: Setting device 1
21:21:16 [DEBUG] (2) gyroflow_core::gpu::wgpu: WGPU adapter (0): AdapterInfo { name: "NVIDIA GeForce RTX 3050", vendor: 4318, device: 9479, device_type: DiscreteGpu, driver: "NVIDIA", driver_info: "531.29", backend: Vulkan }
[... 44 lines cut ...]
21:21:25 [DEBUG] (3) gyroflow::rendering: Possible encoders with Nvidia: [("hevc_nvenc", true), ("hevc_mf", true), ("libx265", false)]
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: HW formats: [CUDA]
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: SW formats: [NV12, YUV420P, YUVA420P, YUV444P, P010LE, P016LE, YUV444P16LE, BGRZ, RGBZ]
21:21:25 [DEBUG] (3) gyroflow::rendering: video_codec: Some("hevc_nvenc"), processing_order: PostConversion
21:21:25 [DEBUG] (3) gyroflow::rendering: start_us: 0, render_duration: 190190, render_frame_count: 5700
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, NV12, P010LE, YUV444P, P016LE, YUV444P16LE, BGRZ, BGRA, RGBZ, RGBA, X2RGB10LE, X2BGR10LE, GBRP, GBRP16LE, CUDA, D3D11]
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12])
21:21:25 [DEBUG] (3) gyroflow_core::stabilization: Setting device 1
21:21:25 [DEBUG] (3) gyroflow_core::gpu::wgpu: WGPU adapter (0): AdapterInfo { name: "NVIDIA GeForce RTX 3050", vendor: 4318, device: 9479, device_type: DiscreteGpu, driver: "NVIDIA", driver_info: "531.29", backend: Vulkan }
21:21:25 [DEBUG] (3) gyroflow_core::gpu::wgpu: WGPU initializing adapter #0
21:21:25 [INFO] Initialized wgpu for (3840, 2160, 3840) -> (3840, 2160, 3840) | key: 1361770713opencv_fisheye80(3840, 2160)(3840, 2160)
21:21:25 [DEBUG] (3) gyroflow_core::stabilization: Setting device 1
21:21:25 [DEBUG] (3) gyroflow_core::gpu::wgpu: WGPU adapter (0): AdapterInfo { name: "NVIDIA GeForce RTX 3050", vendor: 4318, device: 9479, device_type: DiscreteGpu, driver: "NVIDIA", driver_info: "531.29", backend: Vulkan }
21:21:25 [DEBUG] (3) gyroflow_core::gpu::wgpu: WGPU initializing adapter #0
21:21:25 [INFO] Initialized wgpu for (1920, 1080, 3840) -> (1920, 1080, 3840) | key: 1753011054opencv_fisheye80(1920, 1080)(1920, 1080)
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: Some(NV12)
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: JPEG
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_CUDA)
21:21:25 [DEBUG] (3) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x261402a2800
[hevc_nvenc @ 00000260BF8F6BC0] Loaded Nvenc version 12.1
21:21:25 [DEBUG] (3) gyroflow::rendering: [hevc_nvenc @ 00000260BF8F6BC0] Loaded Nvenc version 12.1
[hevc_nvenc @ 00000260BF8F6BC0] OpenEncodeSessionEx failed: out of memory (10): (no details)
21:21:25 [ERROR] [hevc_nvenc @ 00000260BF8F6BC0] OpenEncodeSessionEx failed: out of memory (10): (no details)
```

3dkopfkino, 2023-03-16:
number of render threads 3 --> 4 at 21:21:25

AdrianEddy, 2023-03-17:
That's a bug in ffmpeg, I reported it [here](https://trac.ffmpeg.org/ticket/10268)

It should show an error that your GPU is out of memory instead of crashing, but doing more than 3 concurrent renderings doesn't have any benefits anyway

closed 2023-03-17 by AdrianEddy