gyroflow/gyroflow #616: Can't start Appimage with Ubuntu 22.04LTS

x8n1620, 2023-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

Ubuntu Linux 22.04 LTS 64Bit

### What GPU are you using?

none

### What happened?

The Gyroflow Appimage won't start. Terminal Output is appended.

OS: Ubuntu Linux 22.04 LTS 64Bit, Gnome Desktop, Wayland

Dependencies are installed
sudo apt install libc++-dev libva2 libvdpau1 libasound2 libxkbcommon0 libpulse0 libvulkan1

Non-appimage Gyroflow-linux64.tar.gz runs fine.

### Relevant log output

```shell
# ./Gyroflow-linux64.AppImage
MDK 0.20.0 (git 3dd65ad) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.0 (++20230306093126+ae37edf1486d-1~exp1~20230306213138.48)GCC4.2.1; libc++160000; 01:35:31 Mar  7 2023
15:59:09 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
15:59:09 [DEBUG] (1) gyroflow: Gyroflow 1.5.0 (gh1128)
mdk plugin 'mdk-braw': 0x559584240720 from dir: /tmp/.mount_Gyroflc1xCHg//lib/
mdk plugin 'mdk-r3d': 0x559584241a30 from dir: /tmp/.mount_Gyroflc1xCHg//lib/
/tmp/.mount_Gyroflc1xCHg//gyroflow: symbol lookup error: /lib/x86_64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_proxy_marshal_flags
```

AdrianEddy, 2023-03-29:
Please try latest dev build https://gyroflow.xyz/devbuild/?autodownload

x8n1620, 2023-03-29:
> Please try latest dev build https://gyroflow.xyz/devbuild/?autodownload

Hey @AdrianEddy,
the linked nightly build works fine on my computer.
Thx.

closed 2023-03-29 by AdrianEddy