gyroflow/gyroflow #619: Cannot export video. FFMPEG error with FFMPG installed

graywoulf, 2023-04-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

Windows 10 Pro 64 bit

### What happened?

New to GYROflow and just did a fresh install. Saw that I would need FFMPG to go with it so I downloaded and installed It. Using a RunCam Thumb Pro 4k W with this app. When I try to export a video, I get "ffmpeg::Error(1129203192: Encoder not found)" popup message. I am not sure how FFMPEG works but I have tried running it before starting GYROflow and running it after starting GYROflow. Please tell me what I am doing wrong or have overlooked. Thanks
[image]

### Relevant log output

```shell
I do not know how to do this.
```

AdrianEddy, 2023-04-04:
Your GPU probably doesn't support HEVC encoding. Use H264/AVC or disable GPU encoding

closed 2023-04-04 by AdrianEddy

graywoulf, 2023-04-04:
I am using a Dell 8600 laptop. I cannot find any GPU settings on it. How can I install this H264/AVC encoding?

AdrianEddy, 2023-04-04:
It's in `Export settings` panel on the right, you don't have to install anything. 
One thing is the make sure you have latest GPU drivers installed