gyroflow/gyroflow #620: Auto update feature

IoakeimSogiakas, 2023-04-05:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

First of all, I just want to say that GyroFlow is amazing and now an essential tool in my everyday (I wish) flying!
I have two ideas that I think would be great:
The first which is not that important would be to have a .exe file instead of a zip file to download GyroFlow. 
The second one is more important and probably more helpful. I am suggesting implementing a way to automatically update GyroFlow without deleting the existing files and redownloading the new files. I am no coder so I can't help that much but the least I could do is send this (chat GPTs idea) 

> To implement an automatic update feature in your Windows PC application that is 
located on GitHub, you can follow these general steps:

Set up a version control system: First, you need to have a version control system in place for your project. GitHub provides an excellent platform for version control. Make sure your application code is up-to-date in the repository.

Create a version API: You should create a small API that returns the latest version of the application. This API could be hosted on your website or any other server. The API should return the version number and a link to the latest installer.

Integrate the API in your application: In your application code, you should add a piece of code that queries the API for the latest version number on startup.

Compare the version numbers: After the API query, you can compare the latest version number with the current version of your application. If the latest version is greater than the current version, then you need to update your application.

Download and install the latest version: You should provide a way to download the latest version of your application. You can provide a link to the latest installer on your website or use a third-party library to download the installer from GitHub. After downloading, the installer can be launched automatically to install the latest version.

Notify the user: Finally, you should notify the user that an update has been installed and that they need to restart the application to use the latest version.

Overall, implementing automatic updates in your Windows PC application is not a trivial task, and it requires some coding skills. However, there are several libraries available that can help you achieve this functionality more efficiently.

AdrianEddy, 2023-10-18:
This is now done, Gyroflow is available through all app stores and they handle updates automatically
Windows: https://apps.microsoft.com/store/detail/gyroflow/9NZG7T0JCG9H
macOS and iOS: https://apps.apple.com/us/app/gyroflow/id6447994244
Android: https://play.google.com/store/apps/details?id=xyz.gyroflow

I don't have enough experience with Linux and their package managers, and it's too many different ones to handle. There's #57 request which someone may solve in the future, but at least for now, the .tar.gz will remain recommended on Linux

closed 2023-10-18 by AdrianEddy

IoakeimSogiakas, 2023-10-19:
Amazing man thank you!