gyroflow/gyroflow #621: A flag to not change export settings when uploading a new video

rodrigosclosa, 2023-04-09:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Create a configuration to save the settings from the last export. When we do batch processing, it's very annoying to have to keep reconfiguring all the parameters again for different videos.
We who use it professionally and need to export several cuts of the videos, this functionality will help a lot.

AdrianEddy, 2023-04-09:
You can use `default.gyroflow` preset for that https://docs.gyroflow.xyz/app/advanced-usage/settings-presets#default-preset

rodrigosclosa, 2023-04-09:
> You can use `default.gyroflow` preset for that https://docs.gyroflow.xyz/app/advanced-usage/settings-presets#default-preset

Why not just put a Flag on screen?

rodrigosclosa, 2023-04-09:
During a job where we export several videos in a short period of time, keeping the settings when opening a new video would be the best option. No need to create a preset, because we may have small changes between videos, which we want to keep when loading a new video.
If I have to create a preset every time I adjust the FOV for example, imagine how boring and time consuming it is.

AdrianEddy, 2023-04-09:
The preset can contain only the export settings, which is independent of FOV

The problem with a flag is that it adds complexity to the UI, and since everyone's use case is different, everyone requests a lot of settings tailored to them and it ends up with massive amount of features and options which  have to be carefully designed to not create a mess. It may seem straightforward to add a checkbox but I've been through a lot of these feature requests already and they end up being used by one person

AdrianEddy, 2023-04-09:
At the same time, preset is very flexible and allows you to set whatever you like, you can then set it as default or save in camera_presets directory and have it on the lens profile list in the UI and you can favorite it to be always on top. So you can have any settings you want with 1 click.

That's why preset is preferred over adding any new controls like these to the UI

rodrigosclosa, 2023-04-09:
Not resetting the current screen settings would be excellent. It's very annoying to reset everything when loading a video...

rodrigosclosa, 2023-04-09:
> The preset can contain only the export settings, which is independent of FOV
>
> The problem with a flag is that it adds complexity to the UI, and since everyone's use case is different, everyone [...]

I agree. But you should listen to the use cases of the software to increase it with useful functionalities, for example, for those who use it professionally and need agility when exporting. A voting pool for new features that users want would be great, as there are many user profiles for the software.