gyroflow/gyroflow #623: Being able to choose the part of the video to be exported in vertical resolution

rodrigosclosa, 2023-04-09:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

When exporting vertically, the default is to use the central part of the video... It would be great if we could scroll the video horizontally to better re-frame the video to be exported.

AdrianEddy, 2023-04-09:
You can do that already, under Stabilization -> Advanced -> Zooming center X/Y

closed 2023-04-09 by AdrianEddy

rodrigosclosa, 2023-04-09:
Have you tested it on a real video?
It's not a zoom on the video, it's just moving the X axis. The current functionality gives a horrible zoom on the video, which totally ruins the exported video.

AdrianEddy, 2023-04-09:
Yes i tested and it's exactly for your use case, for vertical videos. It's moving the available rectangle on the X axis, and the stabilization and zooming algorithms adapt to that change so it's better than cropping in post. You can enable Stabilization Overview to see what exactly it does