gyroflow/gyroflow #626: Gyroflow 1.5.0 and nightly Keeps crashing on startup Macosx 12.6.5

riggyfpv, 2023-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

macOS Monterey 12.6.5

### What GPU are you using?

AMD Radeon R9 M395X 4 GB

### What happened?

Gyroflow just keeps crashing at startup after like 1 second. 

### Relevant log output

```shell
22:45:21 [INFO] [MDK] MDK 0.20.0 (git 3dd65ad) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS1090 x86_64; AppleClang14.0.0 (clang-1400.0.29.202)GCC4.2.1; libc++13000; 01:35:52 Mar  7 2023
22:45:21 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
22:45:21 [DEBUG] (1) gyroflow: Gyroflow 1.5.0
22:45:21 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
22:45:21 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
22:45:21 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
22:45:22 [WARN] Populating font family aliases took 147 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
22:45:22 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Apple") Ok("OpenCL 1.2 (Aug  8 2022 21:29:33)") ["AMD Radeon R9 M395X Compute Engine"]
22:45:22 [INFO] OpenCL Platform: Apple, ext: Extensions { inner: "cl_APPLE_SetMemObjectDestructor cl_APPLE_ContextLoggingFunctions cl_APPLE_clut cl_APPLE_query_kernel_names cl_APPLE_gl_sharing cl_khr_gl_event" } Device: AMD AMD Radeon R9 M395X Compute Engine
22:45:22 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_VIDEOTOOLBOX
22:45:22 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_VIDEOTOOLBOX, device: None
22:45:22 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_VIDEOTOOLBOX
22:45:22 [DEBUG] (1) gyroflow::rendering: GPU type: Amd, from name: amd amd radeon r9 m395x compute engine
22:45:22 [INFO] GPU list: ["[OpenCL] Apple AMD Radeon R9 M395X Compute Engine: 1.2", "[wgpu] AMD Radeon R9 M395X (Metal)"]
22:45:23 [INFO] Latest version: v1.5.0, current version: 1.5.0
22:45:23 [INFO] CameraIdentifier {
    brand: "GoPro",
    model: "HERO11 Black Mini",
    lens_model: "",
    lens_info: "Wide",
    focal_length: None,
    camera_setting: "",
    fps: 23976,
    video_width: 2656,
    video_height: 2324,
    additional: "NO-EIS",
    identifier: "gopro-hero11blackmini-wide-2656x2324@23976-no-eis",
}
22:45:23 [DEBUG] (3) telemetry_parser::util: Reading duration: (Some(4.974501196826284), Some(4.974501196826284), None)
22:45:23 [WARN] IMU duration 410368.28709421685 is different than video duration (410368)
22:45:23 [WARN] IMU duration 410411.2724409052 is different than video duration (410368.28709421685)
22:45:23 [INFO] Lens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
22:45:23 [INFO] No gravity vectors - using accelerometer
22:45:23 [INFO] No gravity vectors - using accelerometer
22:45:23 [INFO] Loaded 6390 lens profiles in 203.908ms
22:45:24 [ERROR] thread 'main' panicked at 'already borrowed: BorrowMutError': src/rendering/render_queue.rs:269
   0: __mh_execute_header
   1: __mh_execute_header
   2: __mh_execute_header
   3: __mh_execute_header
   4: __mh_execute_header
   5: __mh_execute_header
   6: __mh_execute_header
   7: __ZN9QtPrivate18QFunctorSlotObjectI12FnBoxWrapperLi0ENS_4ListIJEEEvE4implEiPNS_15QSlotObjectBaseEP7QObjectPPvPb
   8: __ZN9QtPrivate18QFunctorSlotObjectI12FnBoxWrapperLi0ENS_4ListIJEEEvE4implEiPNS_15QSlotObjectBaseEP7QObjectPPvPb
   9: __mh_execute_header
  10: __mh_execute_header
  11: __ZZN21RustQObjectDescriptor8instanceI10RustObjectI7QObjectEEEPKS_vENUlPS2_E_8__invokeES6_
  12: __ZN3QV418QMetaObjectWrapper16virtualIsEqualToEPNS_7ManagedES2_
  13: __ZNK3QV418QMetaObjectWrapper17constructInternalEPKNS_5ValueEi
  14: __ZNK3QV413QObjectMethod12callInternalEPKNS_5ValueES3_i
  15: _qt_v4DebuggerHook
  16: _qt_v4DebuggerHook
  17: __ZN3QV48Function4callEPKNS_5ValueES3_iPNS_16ExecutionContextE
  18: __ZN3QV48Function4callEP7QObjectPPvPK9QMetaTypeiPNS_16ExecutionContextE
  19: __ZN24QQmlJavaScriptExpression8evaluateEPPvPK9QMetaTypei
  20: __ZN25QQmlBoundSignalExpression8evaluateEPPv
  21: __ZN15QQmlBoundSignal10setEnabledEb
  22: __ZN12QQmlNotifier10emitNotifyEP20QQmlNotifierEndpointPPv
  23: __ZN11QMetaObject8activateEP7QObjectPKS_iPPv
  24: __ZN17QQmlVMEMetaObject8metaCallEP7QObjectN11QMetaObject4CallEiPPv
  25: __ZNK11QQmlBinding15hasDependenciesEv
  26: __ZNK11QQmlBinding15hasDependenciesEv
  27: __ZN11QQmlBinding6updateE6QFlagsIN16QQmlPropertyData9WriteFlagEE
  28: __ZN17QQmlObjectCreator8finalizeER26QQmlInstantiationInterrupt
  29: __ZN20QQmlIncubatorPrivate8incubateER26QQmlInstantiationInterrupt
  30: __ZN17QQmlEnginePrivate8incubateER13QQmlIncubatorRK14QQmlRefPointerI15QQmlContextDataE
  31: __ZN13QQmlComponent6createER13QQmlIncubatorP11QQmlContextS3_
  32: __ZN12QQuickLoader9setSourceEP14QQmlV4Function
  33: __ZN12QQuickLoader9setActiveEb
  34: __ZNK11QQmlBinding15hasDependenciesEv
  35: __ZNK11QQmlBinding15hasDependenciesEv
  36: __ZN11QQmlBinding6updateE6QFlagsIN16QQmlPropertyData9WriteFlagEE
  37: __ZN12QQmlNotifier10emitNotifyEP20QQmlNotifierEndpointPPv
  38: __ZN11QMetaObject8activateEP7QObjectPKS_iPPv
  39: __ZN17QQmlVMEMetaObject8metaCallEP7QObjectN11QMetaObject4CallEiPPv
  40: __ZNK11QQmlBinding15hasDependenciesEv
  41: __ZNK11QQmlBinding15hasDependenciesEv
  42: __ZN11QQmlBinding6updateE6QFlagsIN16QQmlPropertyData9WriteFlagEE
  43: __ZN12QQmlNotifier10emitNotifyEP20QQmlNotifierEndpointPPv
  44: __ZN11QMetaObject8activateEP7QObjectPKS_iPPv
  45: __ZN16QQmlAdaptorModel16useImportVersionE13QTypeRevision
  46: __ZN17QQmlDelegateModel15_q_itemsChangedEiiRK5QListIiE
  47: __ZN22QQmlDelegateModelGroup4moveEP14QQmlV4Function
  48: __ZN17QQmlDelegateModel11qt_metacallEN11QMetaObject4CallEiPPv
  49: __ZN11QMetaObject8activateEP7QObjectPKS_iPPv
  50: __ZN18QAbstractItemModel11dataChangedERK11QModelIndexS2_RK5QListIiE
  51: __ZZN3mdk6Player6onSyncENSt3__18functionIFdvEEEiENUlPvE_8__invokeES5_
  52: __mh_execute_header
  53: __mh_execute_header
  54: __ZN7QObject5eventEP6QEvent
  55: __ZN15QGuiApplication5eventEP6QEvent
  56: __ZN23QCoreApplicationPrivate13notify_helperEP7QObjectP6QEvent
  57: __ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
  58: __ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData
  59: _qt_plugin_instance
  60: _qt_plugin_instance
  61: <unknown>
  62: <unknown>
  63: <unknown>
  64: <unknown>
  65: <unknown>
  66: <unknown>
  67: <unknown>
  68: <unknown>
  69: <unknown>
  70: <unknown>
  71: <unknown>
  72: _qt_plugin_instance
  73: __ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE
  74: __ZN16QCoreApplication4execEv
  75: __mh_execute_header
  76: __mh_execute_header
  77: __mh_execute_header

22:45:24 [WARN] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
22:45:24 [ERROR] Object 0x600000b54500 destroyed while one of its QML signal handlers is in progress.
Most likely the object was deleted synchronously (use QObject::deleteLater() instead), or the application is running a nested event loop.
This behavior is NOT supported!
qrc:/src/ui/RenderQueue.qml:373: function() { [native code] }
```

riggyfpv, 2023-04-13:
Apparently don't render files from a removable storage SSD, passed the files over to the internal storage and everything is okay. Also I don't under why my GPU says HEVC is not supported when it should be.

AdrianEddy, 2023-04-18:
This should be fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2023-04-18 by AdrianEddy