gyroflow/gyroflow #628: Gyroflow 1.5.0 and nightly crashes on Ubuntu 22.04 with AMD GPU

x8n1620, 2023-04-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0 

### What operating system are you using?

Ubuntu Linux 22.04 LTS 64Bit

### What GPU are you using?

AMD Radeon RX6600 XT (eGPU in Razor Core v2)

### What happened?

After installing a AMD eGPU on my Ubuntu Notebook Gyroflow crashes at startup
I've tryed stable and nightly
I also tryed the Appimage and the tar.gz Package

### Relevant log output

```shell
#> Gyroflow-linux64/Gyroflow/gyroflow 
0.20.0 (git 7787c8a) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.1 (++20230328073357+42d1b276f779-1~exp1~20230328073502.65)GCC4.2.1; libc++160001; 14:38:49 Apr  9 2023
10:53:46 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
10:53:46 [DEBUG] (1) gyroflow: Gyroflow 1.5.0 (gh1176)
mdk plugin 'mdk-braw': 0x55c31e932380 from dir: /home/xx/Downloads/gyroflow/Gyroflow-linux64/Gyroflow/lib/
mdk plugin 'mdk-r3d': 0x55c31e932fc0 from dir: /home/xx/Downloads/gyroflow/Gyroflow-linux64/Gyroflow/lib/
10:53:47 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 22.2.5") ["AMD Radeon RX 6600 XT (navi23, LLVM 15.0.6, DRM 3.48, 5.19.0-38-generic)"]
10:53:47 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("AMD Accelerated Parallel Processing") Ok("OpenCL 2.1 AMD-APP (3513.0)") []
10:53:47 [INFO] OpenCL Platform: Clover, ext: Extensions { inner: "cl_khr_icd" } Device: AMD AMD Radeon RX 6600 XT (navi23, LLVM 15.0.6, DRM 3.48, 5.19.0-38-generic)
10:53:47 [DEBUG] (1) gyroflow::rendering: GPU type: Amd, from name: amd amd radeon rx 6600 xt (navi23, llvm 15.0.6, drm 3.48, 5.19.0-38-generic)
gyroflow: ../src/wayland-client.c:2312: wl_proxy_set_queue: Zusicherung »proxy->display == queue->display« nicht erfüllt.
Crashdump written to /home/xx/d7f0ab5c-2481-447c-f72b99a4-3ec6c47f.dmp
Abgebrochen (Speicherabzug geschrieben)

Crashdump:
https://www.dropbox.com/s/3vu3iidhi3fuz03/d7f0ab5c-2481-447c-f72b99a4-3ec6c47f.dmp?dl=0
```

AdrianEddy, 2023-04-16:
This looks like a bug in either wayland or Qt, there's not much I can do in Gyroflow about it. You could try setting `export QSG_RHI_BACKEND=vulkan` before running Gyroflow, just to see if using Vulkan works instead of the default OpenGL

x8n1620, 2023-04-17:
After reinstalling the GPU with following options gyroflow will start.
amdgpu-install --usecase=graphics  -y --vulkan=pro,amdvlk --opencl=rocr,legacy --accept-eula

I'm still unable to use GPU-Processing, but this is a problem for anonther topic.

closed 2023-04-17 by x8n1620