gyroflow/gyroflow #631: build error under M1 Mac

harborn, 2023-04-22:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hi, I got error when build under m1 mac, detail info:

command:
```
just run
```

errors info:
```
--- stderr
  thread 'main' panicked at 'Unable to find libclang: "couldn't find any valid shared libraries matching: ['libclang.dylib'], set the `LIBCLANG_PATH` environment variable to a path where one of these files can be found (invalid: [])"', /Users/harborn/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.64.0/./lib.rs:2393:31
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: Recipe `run` failed on line 123 with exit code 101
error: Recipe `run` failed on line 6 with exit code 101
```

and
```
echo $LIBCLANG_PATH
/Library/Developer/CommandLineTools/usr/lib/libclang.dylib
```

AdrianEddy, 2023-04-22:
hmm, what does  `xcode-select --print-path` return?
I believe LIBCLANG_PATH should point to the `lib` directory and not to the libclang.dylib directly

the script assumes the libclang.dylib is located at \`xcode-select --print-path\` + "/Toolchains/XcodeDefault.xctoolchain/usr/lib/"

You can also modify `export LIBCLANG_PATH` in _scripts/common.just

Is this a standard XCode installation or separate command line tools?

AdrianEddy, 2023-05-05:
@harborn did you figure it out? can this issue be closed?

closed 2023-05-14 by AdrianEddy

pacoccino, 2023-10-25:
Just had this issue trying to build gyroflow for the first time. 
I don't have xcode installed, only commandline tools, I had to change 
```export DYLD_FALLBACK_LIBRARY_PATH := if os() == "macos" { `xcode-select --print-path\` + "/Toolchains/XcodeDefault.xctoolchain/usr/lib/" } else { "" }```
to 
```export DYLD_FALLBACK_LIBRARY_PATH := if os() == "macos" { `xcode-select --print-path` + "/usr/lib/" } else { "" }```
in common.just

AdrianEddy, 2023-10-25:
Thanks for the info, this should now be updated in latest commit, please confirm it works for you