gyroflow/gyroflow #632: Default output ratio for 4/3 videos changed to 16/9

pacoccino, 2023-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

3c7125c

### What operating system are you using?

macos 13.3.1

### What GPU are you using?

m1 pro

### What happened?

When I import videos from my DJI Avata recorded in 2.7k 4/3 format (2688x2016), the output ratio is always changed to 16/9 (2688x1512)

For every video I process, I need to click on settings and Original size to keep same size as the input on the output. (sometimes I forget to do so...)

AdrianEddy, 2023-04-23:
This is by design and what most people expect. You can create a default settings preset to override that https://docs.gyroflow.xyz/app/advanced-usage/settings-presets

closed 2023-04-23 by AdrianEddy

pacoccino, 2023-04-24:
Why would most people want to have a different size/ratio on the output than on the input ? If I create a default settings, it'll affect all the video I process, even the 4k 16/9 will be downsized to 2.7 4/3

> This is by design and what most people expect. You can create a default settings preset to override that https://docs.gyroflow.xyz/app/advanced-usage/settings-presets

Why most people would like to have a different size/ratio on the output than on the input ? If I create a default settings, it'll affect all the video I process, even the 4k 16/9 will be downsized to 2.7 4/3

AdrianEddy, 2023-04-24:
> Why most people would like to have a different size/ratio on the output than on the input ?

Load the 4:3 video and turn on the stabilization overview. Then play with Lens correction strength slider. 
You'll notice that the original 4:3 after lens de-stretching actually is much wider than taller, and if you set output side to 4:3, you'll have worst stabilization as Gyroflow will have less room to move around (because the space is now more like 16:9 than 4:3). This happens because of fisheye lens characteristics and by setting the output size to 16:9, you'll be using most possible pixels from your original 4:3 video

pacoccino, 2023-04-24:
I've played around with what you explained and I now understand that there's more room horizontally for stabilization after lens correction (not vertically though ?), that's good to know. 

However I don't think Gyroflow should decide for us to change the aspect ratio, even if it results in smaller space for stabilisation (which just ends up in more zooming ?)

I shoot 2.7k 4:3 videos on the Avata because [its camera gives more vertical field of view](https://youtu.be/MlAMbS4m0ts?t=116) than on 4k 16:9, and it's for editing vertical videos. When gyroflow changes back the ratio to 16:9, I loose the additional FOV I got on 4:3.

I actually stabilized & deleted original footage of a bunch of videos before realizing this behaviour, so these rush are a bit cropped vertically. I may not be the only one who wanted to shoot in 4:3 for editing reel and didn't know gyroflow cropped vertically their video.

AdrianEddy, 2023-04-24:
If you're aiming for vertical videos then you should change export resolution to vertical in export settings, this way Gyroflow will adapt to your available vertical window. You can also keyframe the center offset for to frame your subject if you need. Exporting 4:3 from Gyroflow and cropping to vertical in post will make you lose much more pixels

pacoccino, 2023-04-24:
That's a good option, didn't know I would get less crop by exporting vertical in gyroflow. 

Thank you very much for your responses, and for your amazing software.