gyroflow/gyroflow #634: Empty frozen window on Linux

AdrianRibao, 2023-04-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5

### What operating system are you using?

Arch Linux

### What GPU are you using?

Nvidia

### What happened?

When I try to open gyroflow, the window shows nothing. It's frozen.

I have tried with versions from 1.3 to current dev. Using the tar.gz files, the app images and I have installed the v1.5 from the AUR repository.

I always get the same result.

I think I have all the dependencies installed.

I don't see any error, just the frozen unresponsive window with no content.

Thanks!

AdrianEddy, 2023-04-26:
Launch from the Terminal and let me know the log, or upload `gyroflow.log` file.
Alos you can try a different UI backend: set `export QSG_RHI_BACKEND=vulkan` before launching gyroflow

AdrianRibao, 2023-04-26:
Hi @AdrianEddy this is the output for `cat gyroflow.log`:

```
14:26:29 [INFO] [MDK] MDK 0.20.0 (git 3dd65ad) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.0 (++20230306093126+ae37edf1486d-1~exp1~20230306213138.48)GCC4.2.1; libc++160000; 01:35:31 Mar  7 2023
14:26:29 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
14:26:29 [DEBUG] (1) gyroflow: Gyroflow 1.5.0 (gh1128)
14:26:29 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
14:26:29 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
14:26:29 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
```

after using vulkan it doesn't work either. This is the output:

```
14:26:59 [INFO] [MDK] MDK 0.20.0 (git 3dd65ad) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.0 (++20230306093126+ae37edf1486d-1~exp1~20230306213138.48)GCC4.2.1; libc++160000; 01:35:31 Mar  7 2023
14:26:59 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
14:26:59 [DEBUG] (1) gyroflow: Gyroflow 1.5.0 (gh1128)
14:26:59 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
14:26:59 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
14:26:59 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
```

I am in a desktop computer. I tested it also in the laptop, with an nvidia card and arch linux too, and it worked.

I have tested it with other users also, and didn't work, so it's a system problem.

Thanks for your help!

AdrianRibao, 2023-04-26:
@AdrianEddy `export NO_OPENCL=1` works!!

What a relief, I have been trying lots of things!

Do you know what could be wrong with opencl?

AdrianEddy, 2023-04-26:
Drivers. OpenCL is problematic on Linux, there are many different "adapters" and drivers, but there are so many different distros that I don't know what's the cause exactly. Probably some missing package or a different package for OpenCL is needed. Please google for OpenCL + your GPU + your distro, you'll probably find some info

closed 2023-04-26 by AdrianEddy

AdrianRibao, 2023-04-26:
@AdrianEddy ok, thanks a lot!