gyroflow/gyroflow #636: Crashing at end of render

billjow, 2023-04-28:
Hi, was trying to experiment with using Gyroflow on the output from my new Walksnail HD V2 camera. It got to the end but came up with an ffmpeg error as in the attached screenshot. Gyroflow log attached as well (renamed to gyroflow.txt). I hope you can help! Thanks, Bill

[image]
[gyroflow.txt](https://github.com/gyroflow/gyroflow/files/11354533/gyroflow.txt)

AdrianEddy, 2023-04-28:
Can you send the original video file + gyro data and lens profile?

billjow, 2023-04-29:
Hi Adrian,

Please find below a link to the original video file (too big to email):
[AvatarS0011.txt](https://github.com/gyroflow/gyroflow/files/11359587/AvatarS0011.txt)

[Caddx_Walnut_Wide__1080p_16by9_1920x1080-60.02fps.txt](https://github.com/gyroflow/gyroflow/files/11359590/Caddx_Walnut_Wide__1080p_16by9_1920x1080-60.02fps.txt)

https://1drv.ms/v/s!AvxtLu1q3jWMleIRn9UohbT-Ac5D2A 

Let me know if you have problems getting it.

Please find attached the gyro data (saved as a .txt for uploading) and lens profile (renamed from .json to .txt again) as well.

Please note that I only tried to render one minute of the video file. I also tried both the H265 and H264 encoders, with the sam result.

Cheers, Bill

AdrianEddy, 2023-04-29:
The original video file has some wrong attributes, this is an ffmpeg error and issue with the video encoder in Walksnail. There's nothing I can do in Gyroflow. Trim just a second from the end and it will be working fine.

closed 2023-04-29 by AdrianEddy