gyroflow/gyroflow #638: crash on loading dji action 2 file

gerner, 2023-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

ubuntu 20.04

### What GPU are you using?

AMD Renoir

### What happened?

When opening a video (the first I've shot) from my action 2 gyroflow crashes (see attached log). This is 100% repro with this video file. I've had other video files (e.g. recorded on my goggles) load fine. The video plays fine in VLC.

I tried transcoding the video using ffmpeg (`ffmpeg -i input.mp4 output.mp4`) and gyroflow could open the transcoded file, but it looks like the gyro data didn't make it through the transcoding.

Crashdump: https://drive.google.com/file/d/1Y3mz0uf06EiSbVROn6KWlrBMHBzOnmoL/view?usp=sharing

I'm happy to share the video, but it's going to take a while to upload. I'll add a comment with the link once it's up.

I took a look at the coredump and it looks like this might be in libavcodec/libavutil. So maybe this is an issue with some quirk of this file and those libs?

```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Dwarf Error: DW_FORM_strx1 found in non-DWO CU [in module /home/nick/downloads/Gyroflow/lib/libmdk-braw.so]
Dwarf Error: DW_FORM_strx1 found in non-DWO CU [in module /home/nick/downloads/Gyroflow/lib/libmdk-r3d.so]
--Type <RET> for more, q to quit, c to continue without paging--
Core was generated by `/home/nick/downloads/Gyroflow/gyroflow'.
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0x7fbb537fe700 (LWP 3290271))]
(gdb) where
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007fbca295f859 in __GI_abort () at abort.c:79
#2  0x00007fbca29ca26e in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fbca2af4298 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007fbca29d22fc in malloc_printerr (str=str@entry=0x7fbca2af6ad8 "malloc(): unsorted double linked list corrupted") at malloc.c:5347
#4  0x00007fbca29d52ec in _int_malloc (av=av@entry=0x7fbb3c000020, bytes=bytes@entry=128) at malloc.c:3744
#5  0x00007fbca29d5adf in _int_memalign (av=av@entry=0x7fbb3c000020, alignment=alignment@entry=64, bytes=bytes@entry=24) at malloc.c:4684
#6  0x00007fbca29d99bc in _mid_memalign (address=<optimized out>, bytes=24, alignment=64) at malloc.c:3312
#7  __posix_memalign (size=24, alignment=<optimized out>, memptr=0x7fbb537fd498) at malloc.c:5366
#8  __posix_memalign (memptr=0x7fbb537fd498, alignment=<optimized out>, size=24) at malloc.c:5353
#9  0x00007fbca4e1500d in av_malloc () from /home/nick/downloads/Gyroflow/lib/libavutil.so.58
#10 0x00007fbca4e15146 in av_mallocz () from /home/nick/downloads/Gyroflow/lib/libavutil.so.58
#11 0x00007fbca4dfe063 in ?? () from /home/nick/downloads/Gyroflow/lib/libavutil.so.58
#12 0x00007fbca4dfe008 in av_buffer_create () from /home/nick/downloads/Gyroflow/lib/libavutil.so.58
#13 0x00007fbca4dfe0bc in av_buffer_alloc () from /home/nick/downloads/Gyroflow/lib/libavutil.so.58
#14 0x00007fbca55ce345 in ?? () from /home/nick/downloads/Gyroflow/lib/libavcodec.so.60
#15 0x00007fbca54b33e2 in ?? () from /home/nick/downloads/Gyroflow/lib/libavcodec.so.60
#16 0x00007fbca54b32fb in ?? () from /home/nick/downloads/Gyroflow/lib/libavcodec.so.60
#17 0x00007fbca54b6fb1 in ?? () from /home/nick/downloads/Gyroflow/lib/libavcodec.so.60
#18 0x00007fbca54b4e2b in ?? () from /home/nick/downloads/Gyroflow/lib/libavcodec.so.60
#19 0x00007fbca55ce516 in ?? () from /home/nick/downloads/Gyroflow/lib/libavcodec.so.60
#20 0x00007fbca2c86609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#21 0x00007fbca2a5c133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

I'm happy to help however is useful.

### Relevant log output

```shell
$ ~/downloads/Gyroflow/gyroflow
MDK 0.20.0 (git 3dd65ad) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.0 (++20230306093126+ae37edf1486d-1~exp1~20230306213138.48)GCC4.2.1; libc++160000; 01:35:31 Mar  7 2023
15:20:36 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.2
15:20:36 [DEBUG] (1) gyroflow: Gyroflow 1.5.0 (gh1128)
mdk plugin 'mdk-braw': 0x5577824b42e0 from dir: /home/nick/downloads/Gyroflow/lib/
mdk plugin 'mdk-r3d': 0x5577824b7400 from dir: /home/nick/downloads/Gyroflow/lib/
15:20:36 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 23.0.3 - kisak-mesa PPA") ["RENOIR (renoir, LLVM 15.0.7, DRM 3.42, 5.15.0-71-generic)"]
15:20:36 [INFO] OpenCL Platform: Clover, ext: Extensions { inner: "cl_khr_icd" } Device: AMD RENOIR (renoir, LLVM 15.0.7, DRM 3.42, 5.15.0-71-generic)
15:20:36 [DEBUG] (1) gyroflow::rendering: GPU type: Amd, from name: amd renoir (renoir, llvm 15.0.7, drm 3.42, 5.15.0-71-generic)
15:20:37 [INFO] GPU list: ["[OpenCL] Clover RENOIR (renoir, LLVM 15.0.7, DRM 3.42, 5.15.0-71-generic): 1.1", "[wgpu] AMD Unknown (RADV RENOIR) (Vulkan)", "[wgpu] llvmpipe (LLVM 15.0.7, 256 bits) (Vulkan)", "[wgpu] RENOIR (renoir, LLVM 15.0.7, DRM 3.42, 5.15.0-71-generic) (Gl)"]
15:20:38 [INFO] Lens profiles directory: "/home/nick/downloads/Gyroflow/camera_presets"
15:20:38 [INFO] Loaded 6390 lens profiles in 213.418ms
15:20:39 [INFO] Latest version: v1.5.0, current version: 1.5.0 (gh1128)
15:20:40 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
15:20:42 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/nick/fpv/fomofi/video/first_flight_action2/DJI_0001.MP4"
15:20:42 [DEBUG] (18) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
15:20:42 [DEBUG] (19) MDKPlayer: [in m_player->onEvent] "thread.video" ""
15:20:42 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
15:20:42 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
15:20:42 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
15:20:42 [DEBUG] (1) qml: Setting param smoothness 0.109
15:20:42 [DEBUG] (1) qml: Setting param smoothness_pitch 0.5
15:20:42 [DEBUG] (1) qml: Setting param smoothness_yaw 0.5
15:20:42 [DEBUG] (1) qml: Setting param smoothness_roll 0.5
15:20:42 [DEBUG] (1) qml: Setting param per_axis 0
15:20:42 [DEBUG] (1) qml: Setting param second_pass 1
15:20:42 [DEBUG] (1) qml: Setting param trim_range_only 0
15:20:42 [DEBUG] (1) qml: Setting param max_smoothness 1
15:20:42 [DEBUG] (1) qml: Setting param alpha_0_1s 0.1
15:20:42 [DEBUG] (21) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
15:20:42 [DEBUG] (21) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fbbacf66930) QSize(1218, 685)
15:20:42 [DEBUG] (19) MDKPlayer: [in m_player->onEvent] "decoder.video" "FFmpeg"
15:20:42 [DEBUG] (22) MDKPlayer: [in m_player->onEvent] "decoder.video" "VDPAU"
15:20:42 [DEBUG] (21) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
15:20:42 [INFO] surface size: 3840x2160
15:20:42 [INFO] surface size: 3840x2160
malloc(): unsorted double linked list corrupted
15:20:43 [DEBUG] (21) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 23.0.3 - kisak-mesa PPA") ["RENOIR (renoir, LLVM 15.0.7, DRM 3.42, 5.15.0-71-generic)"]
15:20:43 [INFO] OpenCL Platform: Clover, ext: Extensions { inner: "cl_khr_icd" } Device: AMD RENOIR (renoir, LLVM 15.0.7, DRM 3.42, 5.15.0-71-generic)
15:20:43 [DEBUG] (2) telemetry_parser::dji: Metadata: Object {"clip_meta_header": Object {"proto_file_name": String("dvtm_ac103.proto"), "library_proto_version": String("02.00.01"), "product_proto_version": String("2.0.1"), "product_sn": String(""), "product_firmware_version": String(""), "meta_encryption_type": Number(0), "meta_compression_type": Number(0), "clip_timestamp": Number(29931266), "product_name": String("")}, "clip_streams_meta": Object {"video_stream_num": Number(1), "audio_stream_num": Number(0)}, "distortion_coefficients": Null, "sensor_readout_time": Null, "sensor_read_direction": Null, "digital_focal_length": Null, "eis_status": Object {"status": Number(1)}, "imu_sampling_rate": Null, "sensor_fps": Null}
Crashdump written to /home/nick/fpv/fomofi/video/first_flight_action2/d78c3f7f-676f-43c5-93168784-dc14d0fc.dmp
Aborted (core dumped)
```

gerner, 2023-05-09:
There certainly could have been rocksteady. This is the first video I shot with the camera, and it's mostly on default settings. The video is still uploading. It's 3.6GB. I will disable rocksteady in the future.

But is the crash caused by rocksteady being on?

Thanks a million. The app seems fantastic and I appreciate your work and quick response.

AdrianEddy, 2023-05-09:
Crash shouldn't happen anyway, so I'll let you know once I see the file

gerner, 2023-05-09:
here's the video:

https://drive.google.com/file/d/1z5ab8_WzoqOiSyEuG1zA_hN1dRW381hq/view?usp=sharing

Don't judge my piloting too harshly, I'm just starting out :)

AdrianEddy, 2023-05-09:
hmm, no crash on my side, so it may be something with dependencies. Did you install all recommended packages listed in https://docs.gyroflow.xyz/app/getting-started/installation ?
You could also try setting `export MDK_DECODERS=FFmpeg` before starting gyroflow

gerner, 2023-05-10:
yes, I have all the dependencies and the AMD packages listed.

running `MDK_DECODERS=FFmpeg ~/downloads/Gyroflow/gyroflow` worked and I could load the video! As you mentioned, there's no gyro data in it since it was captured with rocksteady on.

The stacktrace I attached kinda makes it look like it's an issue with the libavcodec or libavutil that's distributed with gyroflow. I have system installed versions of both of those libs, although maybe not the versions that came with my Gyrofllow install. Is that something worth investigating more?

AdrianEddy, 2023-05-28:
The main issue here is listed in the log as `malloc(): unsorted double linked list corrupted` and this looks like some issue inside the GPU driver, VA-API or ffmpeg, and I'm afraid there's nothing I can do from my side. You'll have to stick with just software decoder until there's some driver update or something

closed 2023-05-28 by AdrianEddy