gyroflow/gyroflow #639: Failed to query current video file path.

pikkec, 2023-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

DaVinci Resolve/OpenFX v1.1 for Mac OS

### What operating system are you using?

Ventura 13.3.1

### What GPU are you using?

Macbook m2

### What happened?

I have installed the gyroflow OpenFX plugin for Davinci Resolve 18, Studio version. Davinci finds the app, and I can apply it to clips, but when I click on the 'load for current file button', Davinci displays the following message:

"Failed to query current video file path.
This feature relies on external scripting and is only available in paid Resolve Studio. You have to allow executing scripts:
Set "Preferences -> General -> External scripting using" to "Local".
It must be the currently displayed video on the timeline.
It is also impossible to query file path on a compound clip.
In any case, you can just select the video or project file using the "Browse" button"

I have ensured that all of the issues mentioned in the error message are resolved, and the message keeps on popping up.
I have tried the browse option to add the file in that way and it also doesn't work.

Any help would be appreciated. Thanks.

latenitefilms, 2023-05-10:
Can you share a screenshot?

pikkec, 2023-05-10:
[image]

AdrianEddy, 2023-05-10:
The problem is that MXF files are not sync-less and thus they don't work by themselves in the plugin. You need to synchronize in main Gyroflow app, then export project file and load the project file, or use the "Load for current file" when you have the project file next to the video file. 
Only videos from GoPro Hero 8+ and DJI Avata/O3 Air unit can be stabilized directly without the project file, and they are all MP4, so the plugin only allows loading .mp4 or .gyroflow there

This is not a bug but a lack of documentation which I need to write but is not done yet

latenitefilms, 2023-05-10:
Maybe clicking this button should instead create a new Gyroflow project at the video path and automatically open Gyroflow?

AdrianEddy, 2023-05-10:
maybe it should, good idea

AdrianEddy, 2023-05-26:
latest plugin opens the app when you click "Load for current file" and there is not project and the video file requires synchronization 

https://nightly.link/gyroflow/gyroflow-ofx/workflows/build/main/gyroflow-ofx-windows.zip
https://nightly.link/gyroflow/gyroflow-ofx/workflows/build/main/gyroflow-ofx-linux.zip
https://nightly.link/gyroflow/gyroflow-ofx/workflows/build/main/gyroflow-ofx-macosx.zip
https://gyroflow.xyz/devbuild/?autodownload

Mefisst, 2024-08-09:
I got the same problem with the latest plugin

AdrianEddy, 2024-08-09:
- are you using paid resolve studio?
- did you change the "External scripting" setting?
- Are you using any compound clips?
- Is your timeline playhead over the video you're trying to add the plugin to?

Mefisst, 2024-08-09:
- Yes, I have a paid Studio
- My settings are at default
- I use one 7 second clip
- yes pointer above video

[image]

AdrianEddy, 2024-08-10:
try to reinstall resolve and make sure you have fuscript.exe in Resolve's folder
also, try the latest dev build of the plugin from https://gyroflow.xyz/devbuild/

Mefisst, 2024-09-06:
Removed Davinci and Gyroflow.
Installed it again.
Doesn't work.

AdrianEddy, 2024-09-06:
did you install official gyroflow release, or from the devbuild link I posted? install both dev build of app and plugin

Mefisst, 2024-09-10:
the problem was with the Windows Firewall
it blocked access to the file fuscript.exe