gyroflow/gyroflow #640: Gets error message "Lens profile not loaded" when importing footage from Avata?

DanLeeDK, 2023-05-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

OS Ventura 13.1

### What GPU are you using?

Macbook M1 Max

### What happened?

The program asks me to load a lens profile for my camera. But isn't the Avata camera build into the program? Gyroflow works fine with my GoPro footage.

[image]

AdrianEddy, 2023-05-13:
Make sure you have latest firmware in Avata installed, using DJI Assistant. That's the most common issue. Also make sure you have EIS: Off and FOV: Wide in the camera settings in Goggles.

DanLeeDK, 2023-05-13:
Okay thanks, I'll try that

closed 2023-05-14 by AdrianEddy

bipolartiger, 2024-03-25:
@AdrianEddy Sorry for digging this up, but I don't see anyone talking about this. My Avata's firmware is up to date, EIS: OFF, FOV: Wide. Still says Lens Profile not loaded. And in the search bar, it doesn't show any DJI Avata profile.

[image]

[image]

I was watching a youtuber explaining about Gyroflow and he has the same error, but he just ignore it and doesn't talk about Lens profile. 

The export videos look fine. I can't tell if there is an issue or not with the vid, but I still get the error.

AdrianEddy, 2024-03-25:
The lens profile is loaded on that screenshot, where do you get that message?

dmody, 2025-02-26:
I'm seeing a similar issue.   The gyro data seems to load, but the lens profile says it's not the right lens.   I'm using MACOS M3. 

[image]

AdrianEddy, 2025-02-26:
I've seen that before, you probably had GoPro files previously on the same SD card? what happens is that some gopro data gets leftover on the sd card and it somehow gets to the new DJI file. Then Gyroflow detects the GoPro part instead of DJI

Ideally replace that SD card to a new one. Option 2 is do a full format

Send me your video file and I'll correct it to load DJI

dmody, 2025-03-12:
Thank you for your response.  I have reformatted the card and will do a flight and try again.  In the mean time, I tried accessing a file that was on the internal storage of the Avata 2.  I'm getting the same error.   

Correct me if I'm wrong please, but there is no Avata 2 lens profile that I can load manually?  The A2 lens profile is built into gyroflow, so it's not possible to override and provide it manually?

I tried uploading a 154MB Avata 2 file, but github says it's failed to upload.  I'm not concerned with stabilizing any footage at this moment, just want to get it working for future footage.    

Here is a dropbox link for it though:
[
](https://www.dropbox.com/scl/fi/7i82fe11998y162fydmyg/DJI_20250227154222_0011_D.MP4?rlkey=pnxwd556r85hcu4vulgto7osq&dl=0)

Thanks!!!

dmody, 2025-03-12:
Confirming that I have formatted the SD card and gyroflow is still not loading a lens profile for a newly recorded flight with the avata 2.   Could it be an install problem in MACOS?

[image]

AdrianEddy, 2025-03-12:
looks like you have a default preset installed
1. Go to Advanced on bottom right and "Reset all settings to defaults"
2. Uninstall Gyroflow
3. Install dev build https://gyroflow.xyz/devbuild/
4. In the dev build go to Advanced and "Open data directory" and delete `default.gyroflow` file from there

dmody, 2025-03-12:
I will try what you suggest.  Thank you .  

Just as an aside, I did downloade ver 1.5.4 and it seems to load the lens profile for the Avata.  Not sure what that means, but it's progress, I think.

AdrianEddy, 2025-03-12:
Did you delete the default.gyroflow preset from the gyroflow data directory?
Alternatively, go to Export -> Create settings preset, uncheck everything and select "Save as default preset"

dmody, 2025-03-12:
I believe I didn not delete the default.gyroflow file.  I will try that.  Thank you!