gyroflow/gyroflow #641: DJI Avata file split render missing frame

Mizamook, 2023-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.5.0 (gh1200)

### What operating system are you using?

MacOS M1 Ventura 13.2.1

### What GPU are you using?

OpenCL Apple Apple M1 Max 1.2

### What happened?

Using clips made with the DJI Avata, automatically split at 3.99GB, the crucial part of the clip I like is split.  When I drop the .mp4 files form the Avata into Resolve, they work fine .. the split is frame-accurate.  However, there is at least one frame missing when I process the files using Gyroflow.

Note:  Gyroflow is NOT detecting that there is a file split.  I thought it was supposed to! (Pardon me, it does .. if one selects multiple files on import, and it works perfectly, and renders perfectly .. so my immediate problem is solved)

But I did confirm that Gyroflow is dropping a frame either at the beginning of the render if the split files are rendered separately or at the end.

AdrianEddy, 2023-05-14:
Please share the original video files (not joined)

Mizamook, 2023-05-14:
Will do.  It will take a while (slow internet).

Mizamook, 2023-05-14:
[https://drive.google.com/file/d/1rvjFaDb-wWZ4CA-77YoUfL4krhtQkENT/view?usp=sharing](url)

Here are the files I was having an issue with.  Just for kicks, I'll try it with another set of files (from the same camera) and see if it happens there, too.

AdrianEddy, 2023-05-15:
looks like this is only one file (DJI_0005.MP4)

AdrianEddy, 2023-05-15:
Hmm, I just checked and the DJI_0005.MP4 + DJI_0006.MP4 in Resolve and the DJI_0005_joined.MP4 are exactly identical, frame to frames. But not I re-read your question and it seems that I may misunderstood what you're asking.

You said that there is a frame missing when they are rendered separately, and that would be correct as Gyroflow can drop a frame at the end if it causes an ffmpeg error. But you shouldn't process the files separately anyway, because this breaks stabilization at the split, because the files are treated completely separately. So you should merge first, then stabilize as a single video. Does that work as expected?

Mizamook, 2023-05-15:
Yes, I now know to join the clips (import into Gyroflow together) if I want
the overlap to be merged seamlessly.

And I can confirm it works flawlessly.

Normally I set up shots and execute them, stopping record when I botch
them, reset camera/position, try again.  Usually aiming for 10-30 second
clips.  But learning to fly this thing and hoping for happy accidents
resulted in long files.  Naturally, the one part of the clip(s) I liked
happened to have the file split in it.  Good thing to know now, when it's
not crucial.

So far all other aspects of Gyroflow are brilliant.  Really impressed!

Thank you.

AdrianEddy, 2023-05-15:
Ok so one dropped frame at the very end of the clip is expected, reason for that is some ffmpeg encoder issues. But it doesn't really affect anything when you're processing full merged clip, so I'm closing this issue. 

Gyroflow can't auto-detect split parts from DJI, because there's no filename convention like GoPro does, and there's really no way to tell if the next video is a separate video or continuation. So you have to do it manually, just dragging and dropping 2 (or more) files

closed 2023-05-15 by AdrianEddy